opentx/opentx #6962: Skin of the `X9D+ 2019` Companion simulator

Ceeb182, 2019-10-17:
Currently the skin of the `X9D+ 2019` Companion simulator is the same as the one of the `X9D+`. 
I propose a new skin designed for the `X9D+ 2019` simulator.
[image]

I'm going to create a branch `ceeb182/simuimages` with all the image files on OpenTx repository.

bsongis, 2019-10-17:
I don't like too much this blue :)

Ceeb182, 2019-10-17:
Ok, the color is not pleasing.
What texture or color would you like ?

Ceeb182, 2019-10-18:
Here is a new proposal :
[image]
What do you think about this version ?

Eldenroot, 2019-10-20:
Cool!

3djc, 2019-11-06:
Incorporated

closed 2019-11-06 by 3djc

Ceeb182, 2019-11-07:
The choice of the skin is done. :-)
For more information, see #6963 and #7054.