opentx/opentx #6966: Inputs with stick side POS or NEG not zeroing raw value when crossing sides

elecpower, 2019-10-18:
Steps to reproduce the behavior:
1. Create a new model
2. Create virtual inputs one with LS side=NEG and LS side=POS 
3. Create mixes with these as sources
4. Launch Companion Simulator or Radio
5. Drag LS slider rapidly across the centre point and observe the channel values in the Mixes

Persistent random value is constantly fed into mixes, logical switches, etc when the source is not on the selected side. The raw value is updated when the source returns to the selected side.

From testing the persistent value appears to be related to how quickly the centre point is crossed as the input has random values on either side. This is the difference to #3703 where a 3 position switch was used and only the negative side was observed as being an issue.

Note: the behavior easier to observe in the Simulator on say the sliders and rapidly changing between the positive and negative sides but also observed on the radio hardware.

My use case is for a glider with LS to operate flaps when positive and crow when negative. I have voice callouts at 15s intervals for the current state and this is how I detected the issue.

**Expected behavior**
The raw input value should be zero when not on the selected side and then apply modifiers eg curve.

**General information**
 - OpenTX version 2.2.4
 - Hardware Frsky X9D+
 - OpenTX Companion 2.2.4 on Win 10

kilrah, 2019-10-19:
That's normal, as soon as the input is not in the range when evaluated the line is deactivated. 
If you want to ensure a 0 when the line is deactivated just add an extra line for that.

closed 2019-10-19 by kilrah

elecpower, 2019-10-20:
Thanks @kilrah Set up lines as advised and works as required. I didn't think about the normal logic of the inputs when using a slider and a side.