opentx/opentx #6968: Selection of Different RF Protocol not stored with Companion

Hartinger, 2019-10-19:
OpenTX Version: 2.3.1, OS: Windows 10, Transmitter Type: Horus X10S

I opened an otx file and changed the protocol for a specific model from D16 to D8, then I stored the file and closed the window in Companion.
Upon reopening that same file, I found that the protocol was again set to D16.
I then tried this with different protocol changes and found that in all cases the previously selected protocol was reselected upon reopening the stored otx file.

raphaelcoeffic, 2019-10-19:
Could you please test with the 2.3.2 nightly? I think this has already been fixed since 2.3.1.

Hartinger, 2019-10-19:
I have just tested with the latest 2.3.2 nightly and can confirm that the problem has indeed been fixed.

3djc, 2019-10-19:
Closing then :)

closed 2019-10-19 by 3djc