opentx/opentx #6969: 2.3.2N22 T16 highlighted (bold) display items not visible

Allerhopp, 2019-10-19:
Hi,
little display issue with T16. If items should be displayed in bold letters (highlighted) they are not visible.
[image]

3djc, 2019-10-19:
Not seing that here:

[image]

or you need to be more specific on what your issue is

Allerhopp, 2019-10-19:
If you activate a FM, the according values appear bold. Attached picture is from companion with X10. If you do this with T16, the values become invisible on the radio display. 
[T16_2019_10_19.zip](https://github.com/opentx/opentx/files/3746582/T16_2019_10_19.zip)
[image]
Attached is a model with flightmodes to verify.

3djc, 2019-10-19:
[image]

Taken using your own model on real radio, no issue. Maybe try to reflash + check SDcard

closed 2019-11-06 by 3djc