opentx/opentx #6972: Taranis X9D+ 2019 R9M not entering bind mode

j1and1, 2019-10-19:
So I have updated firmware on the radio and on the R9M module with latest EU firmware.... also the R9 Mini is updated to latest firmware aswell, but when I select R9M EU on my transmitter and enter binding mode it appears that the module is not entering binding mode(no green led and no blinking red led) and I cannot bind my transmitter to my receiver. I can update the firmware for the module using the transmitter, but if I go to the transmitters version screen and click on "Modules" there's no version displayed. 

**Steps to reproduce**
1. Go to model setup page number 2 where the binding happens
2. Scrol down to internal module and turn it off
3. Scroll down to External module and select R9M EU
4. Scroll down and press [Bind]
5. Taranis X9D+ 2019 starts chirping, but the led on R9M stays red(according to it's manual red led should blink and enter binding mode) 
7. Power on the receiver holding down the bind button Green and Red led is lit and its in binding mode
8. No binding happens

**Expected behavior**
R9M enters bind mode and receiver gets bound to R9M in my Taranis X9D+ 2019

**Screenshots**
[image]
[image]

**General information**
 - OpenTX version 2.3.1 (63614a06)
 - Firmware options - crossfire,eu, multimodule, lua, luac

kilrah, 2019-10-19:
What exact firmwares have you loaded to the R9M and R9 Mini

j1and1, 2019-10-19:
R9 Mini currently has firmware with version - 190201 R9_MINI_LBT.frk and the R9M module is flashed with 190117 R9M_LBT.frk
I also have tried the FLEX firmwares on both of them, but still same result.... not binding and no module version displayed.

3djc, 2019-10-19:
As for version of module, I think it is only for ACCESS modules, which yours doesn’t seem to be , so not having version is expected

bsongis, 2019-11-21:
Confirmed. It works on X9D+, but not on X9D+ 2019

j1and1, 2019-11-21:
Good to know that My Taranis X9D+ 2019 is not defective... as I had no other module to test with....

bsongis, 2019-11-21:
I was able to test it on X9D+ (not 2019), same commit

j1and1, 2019-11-21:
> My prototype has inverted TX and RX in the module bay. Would you be able to check if the signal comes from the upper PIN on your radio?

@bsongis  Didn't notice that comment...  Ok will do this when I get home.... Is it as simple as just looking at the markings on the board where module leads are soldered on or do I have to do something else?

MIKE-DAILEY, 2019-11-21:
Tested OK with a X9D Plus SE 2019, R9MFCC and R9.  R9 binds and the channels work.  It takes a bit of time for the R9 to complete the bind process, but it has always worked that way.

bsongis, 2019-11-21:
Thanks Mike, then I can close the issue. @j1and1  Please test with the nightlies!

closed 2019-11-21 by bsongis

j1and1, 2019-11-21:
@bsongis Ok thanks... will do that over weekend.

j1and1, 2019-11-23:
@bsongis So I downloaded the 2.3.3N52 and made sure that the R9M and the R9Mini has the same firmware version.... and still nothing.... the red led on the original R9M module is not blinking to indicate that it is binding that it's supposed to do according to this [manual](https://www.frsky-rc.com/wp-content/uploads/Downloads/Manual/R9M/R9M%20EU%20-manual.pdf) also the green LED on the module is not turning on. So the one thing I have left to try is to get my hands on another radio with a module bay so that I can test the module and to get another compatible module to test with the radio.

Again sorry for the stupid question, but is the Taranis X9D+ 2019 even compatible with the old R9M? 

As the pinout picture shows different pinouts between [R9M 2019](https://www.frsky-rc.com/wp-content/uploads/2019/06/r9m2019decscrip.png) and the [Original R9M](https://www.frsky-rc.com/wp-content/uploads/2017/07/r9mdesp.png). Notice that the top two pins for the R9M 2019 are RX and TX indicating that it is using Serial communication with the module and for the original that I have the top pin is CPPM and one bellow is HeartBeat.

bsongis, 2019-11-24:
Yes they are compatible. Mike did the test as described above. No led at all? Mike, yours is green, right?

j1and1, 2019-11-24:
@bsongis the red LED on the module is lit the green one is off... And once I try to bind there's no LED activity on the module (red -on, green - off)...  I can upgrade the firmware through the taranis tho.... So I really don't know where's the issue at this moment

MIKE-DAILEY, 2019-11-24:
When the system is first powered on the red LED is on not flashing.  In bind mode the green LED and the red LED flashing. When bound the red LED is on and the green LED is mostly on but flickers.  

All works fine for me X9D Plus SE 2019 running OpenTX 2.3.1, with a R9M FCC bound to a  R9.

j1and1, 2019-11-24:
Thanks Mike! Well that's not what's happening to my EU LBT version of R9M.... as when I power on the X9D Plus 2019 the R9M Red LED is on and not flashing and it stays that way permanently.... If I flash the firmware from my X9D Plus 2019 then the green and the red led flashes, but not when I set the X9D to use R9M EU as external module and select the bind option.  R9M is currently flashed with R9M_LBT_20181228 firmware and the firmware version is matched with the RX, but the R9M acts like this despite the firmware it has on.... Also I know that it is producing some kind of RF signal as when I power it on it interferes with my audio system..... So the thing I'll try to do now is to get my hands on another remote with module bay to test the R9M or different module to put in my Taranis.....

MIKE-DAILEY, 2019-11-24:
You could flash the R9M with FCC and see if it works.  No question that FCC works.  

Mike

j1and1, 2019-11-24:
Thanks for suggestion Mike, but it's still not working.... So I'm going to flash everything back and try to get hold on another remote or try to find another way to test the R9M module. Thanks again!

MIKE-DAILEY, 2019-11-24:
It is hard to know if it is a bad R9M or the external module port on your 2019.  You could test the 2019 port with a XJT if you have one.

j1and1, 2019-11-25:
Status update... So I flashed everything back to LBT_20181228 firmware both the RX and the R9M module... Re read the manual and noticed that the module can be paired in the PPM mode.... So I set the dipswitches to 25mW 8ch disconected the module from the Taranis and provided 5V external power while holding the button on the module.... It entered bind mode(green and red LED flashing rapidly)... then I provided the 5V while holding the bind button to the RX and they binded.... 

So I decided to put the module back in the radio and after booting up the module went to its dead state(red LED on not blinking) Note that the Taranis X9D was still set to R9M EU..... So I changed the external module setting to SBUS and the module went to its operational state.... It looked like it was working as the RSSI bar was displayed and it seemed like The radio and the module could communicate.... So everything semi works??? The thing I need to test is if the channel data is present on the RX as I didn't attach the RX to my Quad.

The noob question I have now is: Does this mean that the module port and the module is working? As by my logic SBUS is still used when R9M EU is selected on the X9D+ 2019. Also I assume that the serial communication is restarted or maybe the module is power cycled if the mode is switched  from R9M EU to SBUS as it came to life the moment the mode was changed to SBUS.

kilrah, 2019-11-25:
No, the module doesn't understand SBUS at all.

ST02CMC, 2020-05-11:
> Status update... So I flashed everything back to LBT_20181228 firmware both the RX and the R9M module... Re read the manual and noticed that the module can be paired in the PPM mode.... So I set the [...]

> So I have updated firmware on the radio and on the R9M module with latest EU firmware.... also the R9 Mini is updated to latest firmware aswell, but when I select R9M EU on my transmitter and enter binding mode it appears that the module is not entering binding mode(no green led and no blinking red led) and I cannot bind my transmitter to my receiver. I can update the firmware for the module using the transmitter, but if I go to the transmitters version screen and click on "Modules" there's no version displayed.
> 
> **Steps to reproduce**
> 
> 1. Go to model setup page number 2 where the binding happens
> 2. Scrol down to internal module and turn it off
> 3. Scroll down to External module and select R9M EU
> 4. Scroll down and press [Bind]
> 5. Taranis X9D+ 2019 starts chirping, but the led on R9M stays red(according to it's manual red led should blink and enter binding mode)
> 6. Power on the receiver holding down the bind button Green and Red led is lit and its in binding mode
> 7. No binding happens
> 
> **Expected behavior**
> R9M enters bind mode and receiver gets bound to R9M in my Taranis X9D+ 2019
> 
> **Screenshots**
> [image]
> [image]
> 
> **General information**
> 
> * OpenTX version 2.3.1 ([63614a0](https://github.com/opentx/opentx/commit/63614a06f81c910b949a918ef89b40d1df241aaf))
> * Firmware options - crossfire,eu, multimodule, lua, luac

HI

Wondering if you got to the bottom of this issue, I have 2 x Taranis 1 X9D+ and 1 x X9D+ 2019, both have R9 Modules and the X9D+2019 will not bind via the R9M module.

I have switched the Modules between both Taranis and the X9D+2019 still has issue Binding via the R9M, the X9D+ has no issue bind with either module in use.

Both Taranis are running on the Same firmware 2.3.7 and Both R9Modules have the same latest Flex Firmware on them.

1 .  X9D+ / Opentx 2.3.7 / both R9Modules Bind with the R9Slim+ no issues
2.   X9D+2019 / Opentx 2.3.7 both R9Modules will not bind with the R9 Slim+

I have tried rolling back the Firmware to an Earlier version of both Opentx and FrSky but still cannot get either module to work in the 2019 X9D+

If anyone has any ideas I would really Appreciate them

thanks

j1and1, 2020-05-12:
@ST02CMC No I didn't figure out exactly why it does not bind when using R9M in X9D+2019. Taranis can update the module firmware, it can receive telemetry data while bind in PWM mode, but for some reason the communication just does not work with the standart R9M Module(there's no activity indicated by the leds on the module) :(

ST02CMC, 2020-05-12:
> @ST02CMC No I didn't figure out exactly why it does not bind when using R9M in X9D+2019. Taranis can update the module firmware, it can receive telemetry data while bind in PWM mode, but for some [...]

Hi

Thanks for coming back to me, It is driving me up the wall as it don`t make any sense at all.

Sceptorrh, 2020-05-17:
Same exact problem over here, using the non 2019 x9d+ se and the 2019 version of the r9m module. The module won't go into binding mode...

KenMurphy, 2020-06-23:
Sorry to awaken this closed thread, but I had the same issue with my R9M 2019 and X9D+ 2019. I took the radio apart and found something that may be of interest. I posted the details here on RCGroups: https://www.rcgroups.com/forums/showthread.php?3639129-X9D-2019-not-working-with-R9M-2019

FrSky has been extremely slow to respond to my emails and so far has basically just ignored what I said and told me to make sure I was using the correct firmware, which I 100% am.

Unb9rn, 2020-07-11:
Were anyone able to solve this problem? I am having exactly the same issue, X9D+ 2019 with R9M 2019,-I am able to update external module with any firmware. External module version is not displayed in module info, it won't bind at all, the LED always remains red. 
@j1and1 @KenMurphy @Sceptorrh have you contacted frsky? is this case covered by warranty?

alm-nl, 2020-07-11:
> Were anyone able to solve this problem? I am having exactly the same issue, X9D+ 2019 with R9M 2019,-I am able to update external module with any firmware. External module version is not displayed [...]

Take a look at the topic at rcgroups as mentioned above by KenMurphy. You could check if you have the same wire between the pins or measure if there is a short between them (upper two pins).

KenMurphy, 2020-07-11:
@Unb9rn I wasn't able to solve it with my original backboard unfortunately. I replaced the backboard and everything works as expected. I would start with what @alm-nl suggested, check for continuity between the pins and inspect the backboard. The RCGroups link above has all the details about my module & what I found if you want to compare, I kind of just used that thread as a journal after FrSky didn't seem interested in helping in a reasonable manner.