opentx/opentx #6975: [2.3.1] - Custom failsafe conflict between internal and external module

robustini, 2019-10-20:
On FrSky Taranis X9D+ with OpenTX V2.3.1 I installed an external R9M 2019 module with Flex firmware.
I'm noticing that the custom failsafe on the external module are neither written nor respected, it seems that even when entering to set them up it takes those of the internal module.

**To Reproduce**
1: on XJT internal module in D16 mode set a custom failsafe value in all the 16 channel
2: disable the internal module
3: enable the external module R9M, try to set a custom failsafe value, and exit
4: return to the custom failsafe value, they will not be the ones you wrote

**General information**
 - OpenTX: V2.3.1 (same problem with 2.2.4)
 - Hardware: Frsky X9D+
 - OpenTX Companion: V2.3.1

3djc, 2019-10-20:
There is also an inconsistency I think between radio and companion data models for this

robustini, 2019-10-20:
> 
> 
> There is also an inconsistency I think between radio and companion data models for this

The absurd thing is that in the Companion simulator it seems to work instead.

3djc, 2019-10-20:
We probably have not done a great job (and yes, that is an understatement) in communicating a crucial change in 2.3. We are working on that

kilrah, 2019-10-20:
There is only one failsafe value per channel, because there is no reason you would send the same channel out to 2 modules but wanting different failsafe positions for each.

robustini, 2019-10-20:
> 
> 
> There is only one failsafe value per channel, because there is no reason you would send the same channel out to 2 modules but wanting different failsafe positions for each. So on the radio both failsafe settings pages show the same thing.
> Seems companion has an issue indeed.

Absolutely correct, the custom failsafe must be distinguished in both.

kilrah, 2019-10-20:
No, companion also has to use the same shared values.

robustini, 2019-10-20:
Not good.

3djc, 2019-10-20:
If you think that, you will need to elaborate more on use case where that would be needed

robustini, 2019-10-20:
In the view that both modules act on the same system while flying I agree with you.

kilrah, 2019-10-20:
No other option. And having one set of failsafes was a deliberate decision based on what I mentioned in my previous post.

robustini, 2019-10-20:
But one module could for example pilot a drone, and the other a second rx connected to a gimbal.
In that state the custom failsafe should be distinguished from each other.

kilrah, 2019-10-20:
No! You'll be using different channels, so it's already ok.

robustini, 2019-10-20:
Ok, i agree about that.
But the problem written above remains, please check.

3djc, 2019-10-20:
I cannot reproduce you issue. No matter how modules are setup, I always have the right commun failsafe values

robustini, 2019-10-20:
> 
> 
> I cannot reproduce you issue. No matter how modules are setup, I always have the right commun failsafe values.

Leaving the internal module disabled, try to change a custom value on the second, exit and go back inside, here it remains as before.

3djc, 2019-10-20:
Yes, still can't reproduce :s

robustini, 2019-10-20:
This does not happen in the simulator.

3djc, 2019-10-20:
Don't have a x9d(+) closes I have is X9E, and can't reproduce with it either

robustini, 2019-10-20:
Perhaps in the X9D it does the same thing, if you want to try this it is the otx, model "GAIA DN".

[da.zip](https://github.com/opentx/opentx/files/3747696/da.zip)

3djc, 2019-10-20:
Dont have either an X9D or X9D+, so if it not the case on simu, somebody else will  have to take it

robustini, 2019-10-20:
Question: when the rx is turned on and i try to set a costom value, when i enter inside "SET" the tx read this value from the rx turned on?
I ask this because now I am trying with the rx off and on the TX it keeps them.

3djc, 2019-10-20:
No, not in ACCST

robustini, 2019-10-20:
Ok, i use R9M 2019 in FLEX here.

3djc, 2019-10-20:
So ACCST, so no, we can't read data from ACCST receivers

robustini, 2019-10-20:
Nice, then the problem it seems is with the R9M in FLEX.
We have tried now to turn on the rx and in fact it does not keep the ones I set.

3djc, 2019-10-20:
Trying to recap.
X9D, internal module disabled, r9m flex enabled, rx off : you can change custom failsafe and values are saved

Same setup with rx on, it does not. Is that what you are saying ?

robustini, 2019-10-20:
> 
> 
> Trying to recap.
> X9D, internal module disabled, r9m flex enabled, rx off : you can change custom failsafe and values are saved
> 
> Same setup with rx on, it does not. Is that what you are saying ?

Correct! If the rx is turned on does not save them.

robustini, 2019-10-20:
I have no words, now he does so with the rx on, I close, thank you.

closed 2019-10-20 by robustini