opentx/opentx #6989: OpenTX 2.3.2.N25 issue with external module simulator for T16

backman8571, 2019-10-23:
I installed N25 load and it only gives me 6 channels for DSM/X for T16

1 open the radio simulator

2, select external model

3. select DSM and D11/22 or X11/22

4. go to select channel range it will only allow upto 6 channels

**Expected behavior**
The radio should allow 12 or 16 channels

**General information**
 - OpenTX version2.3.2 N25
 - Hardware Jumper T16Pro Plus
 - OpenTX Companion on Windows 10

3djc, 2019-10-23:
Are you talking on the radio ? simu ? companion ? I get 12 on the radio as shown bellow

[image]

MRC3742, 2019-10-23:
Are you possibly selecting "DSM2" as protocol instead of "MULTI" and then "DSM" as sub protocol?

DSM2 Protocol is for the earlier DSM Hack module and only had six channels available.

MRC3742, 2019-10-24:
@3djc  The problem is definitely in Companion 2.3.2N25 - T16

Setting the DIY Multiprotocol Module on either Internal or External and Protocol to DSM, any of the sub protocol choices originally show "6" under channels value . 
This channel value can only be changed from 6 to a value of 4, 5, or 6.

Using the radio it works properly.
When DSM is selected, Channel Range shows CH1-CH7 and now can be changed to any value from CH4 to CH12 .

3djc, 2019-10-24:
Thanks @MRC3742 , with a proper description of the issue, i was able to fix it :)

PR #6997 (2019-10-24): Fix multi dsm channel count/max
- max chan for multi dsm is 12 (as oposed to dsm hack module : 6)
- on multi change, set channelCount to max (was already done for non multi protocol change)

This fixes #6989
closed 2019-10-24 by bsongis