opentx/opentx #6994: 2:3:1 X10 Express: external antenna selection not working properly.

eovnu87435ds, 2019-10-23:
I think there's an issue with using the external antenna with D16 mode on the ISRM Module in X10 Express. 

If I put a 2.4GHz Yagi on the X10 express, set the module settings to ISRM, D16, and external antenna checked, I can put the radio in range test, stand 20m from the receiver, and no matter where I point the yagi, the RSSI is approximately the same (around 60). If I unscrew the antenna, I do get the antenna defective warning.

If I do the same exact test with an X10s in D16 mode, (running OpenTX 2.2.3), when I do range testing I get drastically different results depending on whether or not the yagi is pointed at the receiver. 

Thus far, I have tested this on 3 X10 radios and 3 X10 express radios and they all operated the same.  I've also tested it in flight and found that the X10 radios are properly using the directional and can get a longer range when the antenna is aimed properly. The Express radios have exactly the same performance as when external antenna was un-checked.

robustini, 2019-10-24:
Now with the absence of the RAS data it's no easy to understand which antenna is working and how.

raphaelcoeffic, 2019-10-25:
Quick question: did you use the « options » that leads to a different screen, or something else?

bsongis, 2019-10-27:
It is not related to opentx, is it?

bsongis, 2019-10-28:
There is a difference between X10S and X10S Express:

When you select External Antenna active on X10S Express, then both antennas are active (time sharing), while on X10 the internal antenna was completely disabled.

closed 2019-11-21 by bsongis