opentx/opentx #6995: Companion 2.3.1 logical switch editing bug

Miami-Mike, 2019-10-23:
There's a logical switch editing bug in Companion version 2.3.1 that's not in version 2.2.4.

Create two logical switches as L1 and L2, where V1 and/or V2 of L1 point to L2. L2 can be anything. 

For example:

`L1   SA↑ AND L02`
`L2   S1 > 0`

Now either cut and paste, or copy and paste, L2 to L3, which is something you might do to make space for an added switch between those two switches.

`L1   SA↑ AND L02`
`L2 `
`L3   S1 > 0`

Because you moved L2 to position L3, you'll now want to change the V1 and/or V2 pointers of L1 to point to L3 instead of L2. In Companion 2.2.4 you'll be able to do that because L3 will now show up as a choice, but in Companion 2.3.1 the choice of L3 does not show up.

A workaround to get L3 to show up as a choice is to create a dummy logical switch as L4. Just choosing a function for L4 alone is sufficient to get L3 to show up as a choice for L1.

Miami-Mike

kilrah, 2019-10-24:
Can you see if L3 becomes a choice if you close/reopen the window after the cut/paste?

Miami-Mike, 2019-10-24:
Yes, clicking to another page and then back to the Logical Switches page causes the choices to display properly.

PR #7106 (2019-11-21): Fix bug in cpn LS edition
This fixes #6995
commit a64010d0 (2019-11-21): Fix #6995 (#7106)
closed 2019-11-21 by bsongis