opentx/opentx #7004: [2.3.2 DEV - Spektrum analyzer issue (ISRM)

robustini, 2019-10-25:
@3djc I'm experiencing a strange behavior with the spektrum analyzer on Horus X10S Express ACCESS (ISRM), I'm attaching a video to ask you if this very slow scroll is normal.
The comparison with the Multiprotocol is tangible (external in my case).

https://youtu.be/ZDNMAWAdU5s

**General information**
 - OpenTX: V2.3.2 DEV (commit 1b35f2b)
 - Hardware: Horus X10S Express ACCESS + Multiprotocol external module

3djc, 2019-10-25:
Unfortunately, I do not have any equipement that can help diagnose that. There seem to be 2 issues here, the value seems wrong, and the slow refresh. We will need to find somebody with the right gear and skills to investigate

robustini, 2019-10-25:
I honestly think I'll never use it, it was just to know, thanks!

robustini, 2019-10-25:
If you take a good look at the video there is also a graphic problem when INT is selected, the peak of the bars goes to cover the line with the frequencies, with the EXT it does not.

kilrah, 2019-10-25:
Yup we know, saw that yesterday

raphaelcoeffic, 2019-10-25:
It’s the same at my place, I will probably take over

Eldenroot, 2019-10-27:
Is that fixable?

robustini, 2019-10-29:
> 
> 
> Is that fixable?

I hope! :p

bsongis, 2019-11-08:
aie aie aie

robustini, 2019-11-09:
@bsongis sure it's okay?
I tried now and it got worse because now it does the same thing with the MPM too.
The refresh is about 2 Hz, both INT and EXT.

bsongis, 2019-11-09:
I tested on the internal module on X10 Express. There were a lot of issues. Do you confirm it's OK there?

3djc, 2019-11-09:
Bertrand fixed the display glitches, not the refresh rate. As for external MPM refresh rate, what is your module version ?

3djc, 2019-11-09:
@robustini tested on T16 EXT, I had 1.3.0.33 installed and that was slow. Tried 1.3.0.22 and 1.2.1.85, both work ok here

robustini, 2019-11-09:
> 
> 
> @robustini tested on T16 EXT, I had 1.3.0.33 installed and that was slow. Tried 1.3.0.22 and 1.2.1.85, both work ok here

Ok, has been fixed only the glitches.
Tested with MPM 1.3.0.40.

3djc, 2019-11-09:
Could you try with 1.2.1.85 and confirm speed is ok ?

robustini, 2019-11-09:
Tested, i can confirm.
But the problem of refreshing on the ISRM remains.

bsongis, 2019-11-09:
On ISRM is there a version which has been better?

robustini, 2019-11-09:
> 
> 
> On ISRM is there a version which has been better?

No, he always did this thing.

commit afeca718 (2019-11-08): Fixes #7004
closed 2020-08-05 by stale[bot]