opentx/opentx #7005: Jumper T16 betaflight-tx-lua-scripts not working

tiaanv, 2019-10-25:
The betaflight TX lua script not working as expected.

- Jumper T16 Pro
- BF script 1.4.1
- OpenTX 2.3.2N27 nightly build
- Multiprotocol module updated to latest with bootloader (FWIW)

The BF.lua script runs fine... 
Can also page through various settings... but nothing has values.
Telemetry works fine as can be seen below:
[image]

BF script on my Taranis X7 with exact same quad worked fine...
Logged issue on betaflight-tx-lua-scripts Github also
https://github.com/betaflight/betaflight-tx-lua-scripts/issues/272

3djc, 2019-10-25:
No need to log them at beta flight. It is a known multi/OpenTX issue. Fix is planned

tiaanv, 2019-10-25:
> No need to log them at beta flight. It is a known multi/OpenTX issue. Fix is planned.
> 
> https://youtu.be/eCnxs3d4VKg

Awesome. Thx.

Harrykd33, 2019-11-06:
Having a similar issue with brain2 integration, im not getting any values from the Brain unit.
Lua file can be found here under frsky integration. https://www.msh-electronics.com/downloads/

bsongis, 2019-11-06:
@Harrykd33 not at all the same issue, here is discussed an issue with MULTI (polling not available)

Harrykd33, 2019-11-07:
Yes correct that's the exact issue I'm having on my jumper T16pro with the internal multi module. It's not pulling bidirectional telemetry. Without bi telemetry my lua script does not work.

kilrah, 2019-11-07:
Yes it's known that no uplink works with multi at that point, you have to wait until it's implemented.

3djc, 2019-11-18:
Fixed in N45

closed 2019-11-18 by 3djc

tiaanv, 2019-11-18:
I've just updated to N46.  I still have no joy.  No BF values in BF.lua
Notes:
* Under the Model Setup - the Module Status says "Module Upgrade Required".

Do I really have to take apart my TX again to flash the multiprotocol module with an fdti adapter again? I thought that after the initial re-flash it should have the bootloader on now to flash form TX..... 

If this post is no longer relevant here, someone can direct me where I can perhaps get some help.

3djc, 2019-11-18:
Yes, if you have installed the bootloader on your multi, you should be able to flash it from the radio. You need inv if you are using external, noinv for internal. Put the multi firmware on your sd, run the sd navigator, long enter, and choose flash multi

Tested successfully, and others have confirmed it working (the BF scripts)

tiaanv, 2019-11-18:
Thx @3djc. I was being impatient, and quite new to the T16.  It can't believe it was as simple as navigating to the BIN file on the SD and selecting "Flash Internal Multi".

I can confirm it is working as Expected!!! Yah!!!