opentx/opentx #7006: Building OpenTX under Windows

SHezinger, 2019-10-25:
Building firmware (X12S) under Windows fails. It seems that there are some libs missing
Building companion works fine.
Any ideas?

`FAILED: radio/src/targets/common/arm/stm32/bootloader/bootloader.elf 
cmd.exe /C "cd . && arm-none-eabi-g++ -mcpu=cortex-m4 -mthumb -fomit-frame-pointer -fverbose-asm -Wa,-ahlms=firmware.lst -Os -gdwarf-2 -DHSE_VALUE=12000000 -fno-exceptions -fdata-sections -ffunction-sections  -Wall -Wno-address-of-packed-member -Wno-strict-aliasing -Wformat -Wreturn-type -Wunused -Wuninitialized -Wunknown-pragmas -Wno-switch -Wtype-limits -mfloat-abi=hard -mfpu=fpv4-sp-d16 -mcpu=cortex-m4 -mthumb -nostartfiles -lm -TC:/Users/SHezinger/Desktop/Software/opentx/radio/src/targets/horus/stm32_ramboot.ld -Wl,-Map=bootloader.map,--cref,--no-warn-mismatch,--gc-sections --specs=nano.specs @CMakeFiles\bootloader.rsp  -o radio\src\targets\common\arm\stm32\bootloader\bootloader.elf && cmd.exe /C "cd /D C:\Users\SHezinger\Desktop\Software\build-opentx-Desktop_Qt_5_13_1_MinGW_32_bit-Release\radio\src\targets\common\arm\stm32\bootloader && arm-none-eabi-objcopy -O binary bootloader.elf bootloader.bin && cd /D C:\Users\SHezinger\Desktop\Software\build-opentx-Desktop_Qt_5_13_1_MinGW_32_bit-Release\radio\src\targets\common\arm\stm32\bootloader && C:\Users\SHezinger\AppData\Local\Programs\Python\Python38-32\python.exe C:/Users/SHezinger/Desktop/Software/opentx/radio/util/bin2lbm.py bootloader.bin bootloader.lbm""
c:/program files (x86)/gnu tools arm embedded/4.7 2014q2/bin/../lib/gcc/arm-none-eabi/4.7.4/../../../../arm-none-eabi/bin/ld.exe: cannot find -lkernel32
c:/program files (x86)/gnu tools arm embedded/4.7 2014q2/bin/../lib/gcc/arm-none-eabi/4.7.4/../../../../arm-none-eabi/bin/ld.exe: cannot find -luser32
c:/program files (x86)/gnu tools arm embedded/4.7 2014q2/bin/../lib/gcc/arm-none-eabi/4.7.4/../../../../arm-none-eabi/bin/ld.exe: cannot find -lgdi32
c:/program files (x86)/gnu tools arm embedded/4.7 2014q2/bin/../lib/gcc/arm-none-eabi/4.7.4/../../../../arm-none-eabi/bin/ld.exe: cannot find -lwinspool
c:/program files (x86)/gnu tools arm embedded/4.7 2014q2/bin/../lib/gcc/arm-none-eabi/4.7.4/../../../../arm-none-eabi/bin/ld.exe: cannot find -lshell32
c:/program files (x86)/gnu tools arm embedded/4.7 2014q2/bin/../lib/gcc/arm-none-eabi/4.7.4/../../../../arm-none-eabi/bin/ld.exe: cannot find -lole32
c:/program files (x86)/gnu tools arm embedded/4.7 2014q2/bin/../lib/gcc/arm-none-eabi/4.7.4/../../../../arm-none-eabi/bin/ld.exe: cannot find -loleaut32
c:/program files (x86)/gnu tools arm embedded/4.7 2014q2/bin/../lib/gcc/arm-none-eabi/4.7.4/../../../../arm-none-eabi/bin/ld.exe: cannot find -luuid
c:/program files (x86)/gnu tools arm embedded/4.7 2014q2/bin/../lib/gcc/arm-none-eabi/4.7.4/../../../../arm-none-eabi/bin/ld.exe: cannot find -lcomdlg32
c:/program files (x86)/gnu tools arm embedded/4.7 2014q2/bin/../lib/gcc/arm-none-eabi/4.7.4/../../../../arm-none-eabi/bin/ld.exe: cannot find -ladvapi32
collect2.exe: error: ld returned 1 exit status
[89/341 2.3/sec] Building CXX object companion/src/CMakeFiles/simulator.dir/simulator.cpp.obj
ninja: build stopped: subcommand failed.
16:49:58: Der Prozess "C:\Program Files\CMake\bin\cmake.exe" wurde mit dem Rückgabewert 1 beendet.
Fehler beim Erstellen/Deployment des Projekts OpenTX (Kit: Desktop Qt 5.13.1 MinGW 32-bit)
Bei der Ausführung von Schritt "Erstellen mit CMake"`

kilrah, 2019-10-25:
This is for reporting and tracking bugs, not discussion. Besides you say nothing of your environment, which of the wiki methods you followed, what exactly fails ect...

closed 2019-10-25 by kilrah

SHezinger, 2019-10-25:
Sorry, but I thought to create a help issue is the right thing to do. Where should I ask instead? Thanks