opentx/opentx #7007: Companion 2.3: Import model adapter 2.2 to 2.3?

afdofdez, 2019-10-25:
In windows 10, the simulator 2.3.1 with models imported from version 2.2 continuously gives the error "low transmitter battery", with models created in version 2.3.1 works correctly.

3djc, 2019-10-26:
What are your battery alarm values in both cases ?

afdofdez, 2019-10-26:
In model 2.2 battery warning 11,1 v (battery range 10,5 – 12,6)
In model 2.3 battery warning 6,6 v (battery range 6,0 – 8,0)

3djc, 2019-10-26:
The stock battery for X9D+ is a Ni-MH 7.2V 2000mAh battery, If you have those value in 2.2, they are not default values and you probably did some mod on your radio for a 3S lipo (guessing here), and adjusted radio setting accordingly

afdofdez, 2019-10-26:
Conclusion if a different battery voltage than standard is set in version 2.3, the radio simulator gives transmitter battery error.  It is still an error because if the battery and the warning level are configurable, it should not fail to configure values ​​other than the standard.

3djc, 2019-10-26:
Well no, 2.2 and 2.3 set the same standard 6,6 v (battery range 6,0 – 8,0). You say your model have 11,1 v (battery range 10,5 – 12,6), but that's no standard value

3djc, 2019-10-26:
2.2 default:
[image]

2.3 default:
[image]

3djc, 2019-10-29:
Closed, replaced by new feature request : #7019

closed 2019-10-29 by 3djc