opentx/opentx #7008: T16 / X10 - unable to select custom wav files as background music

jeffhendrix, 2019-10-25:
On a T16 radio and in companion for T16 and X10, running release 2.3.2N27.
Unable to select custom wav files for background music.
These files work fine on my X9D.
Trying to select them on a T16 radio, they don't show up in the drop down.
Same things happens in companion for both T16 and X10.
Setting companion to X9D profile, the files show up in the drop down.
Selecting the wav files from the SD browser, will play them correctly, both on the radio and in the simulator.

3djc, 2019-10-26:
Is your wav files 6chars or less and located at the right place ?

kilrah, 2019-10-26:
Did you configure the SD card path for your new T16 profile?

jeffhendrix, 2019-10-27:
That problem was the file length. Making them 6 characters or less fixed the problem.
So why can I use longer file names on my X9D, but the T16 is limited to 6?

closed 2019-10-27 by jeffhendrix

notlebowski, 2019-10-27:
https://github.com/opentx/opentx/issues/6198

notlebowski, 2019-10-27:
I think it's confusing to allow more than 6-chars on the X9D whilst other models have that limit.

it's also not documented at all. on the contrary. documentation even explicitly says it's firmware version dependent