opentx/opentx #7010: Telemetry mirror and bluetooh

Guillesan, 2019-10-26:
I am following the development for Jumper T16. My request is if you can have "mirror telemetry", either by physical port, or by a bluetooh module. Right now I have a bluetooh module installed for Horus 10 on my Jumper T16, the visibility of the bluetooh works, it is even possible to change the name but the telemetry does not come out. Greetings

3djc, 2019-10-26:
Are you using self compiled binaries ?

Guillesan, 2019-10-26:
I tested with the Jumper T16 2.3 firmware and with the latest night version

3djc, 2019-10-26:
Then they do not support BT at all

Guillesan, 2019-10-26:
That is a problem, it could be overcome. In my other Taranis 9XE station I have a Wi-Fi module installed in the physical serial port, it works very well, I would like to do the same in my Jumper T16

robustini, 2019-10-26:
Bluetooth on T16 works really well, but the automatically compiled firmware does not provide it, so at the moment you have to compile it with the BLUETOOTH option.

redjacketKR, 2019-10-27:
> I am following the development for Jumper T16. My request is if you can have "mirror telemetry", either by physical port, or by a bluetooh module. Right now I have a bluetooh module installed for [...]

I totally agree that. I don't know all that inversion / noinversion telemetry stuff yet but

1) people are using extra sport of battery bay from 'taranis x9d' using a little inversion hardware
    T16 supposedly has one unused uart so it's possible to pull out gps info via UART or internal BT module, theoretically (for those who use Micro XF)

2) Or you can also use that as trainer with spektrum or taranis etc..

3djc, 2019-10-27:
1)  support for that will come in time, we are just very busy on a lot of things right now

2) there is a much better option on T16 with multi trainer, very very sweet

Guillesan, 2019-10-27:
Hello everyone. As I am told, a special compilation is needed, I am not in a position to do so that I surpassed today. 3djc you tell me that from now on it will be possible to be implemented, I will wait, I am very interested in it, telemetry in my case be prepared from its normal screen use of the Jumper T16 in my case I would also use it in our tracker antenna project www.u360gts. com

nelsong, 2019-10-28:
> Bluetooth on T16 works really well, I use that of the X12S (the most economic, $9), but the automatically compiled firmware does not provide it, so at the moment you have to use a self compiled binaries with the BLUETOOTH option.

Can you share the flags you used to compile it? I compiled mine with:
```shell
docker run -v /root/opentx:/opentx -e "OPENTX_VERSION_SUFFIX=2.3" -e "BLUETOOTH=yes" -w /opentx/build-t16 opentx cmake -DPCB=X10 -DGVARS=YES -DLUA=YES -DLUA_COMPILER=YES -DHELI=NO -DDEBUG=NO -DCMAKE_BUILD_TYPE=Release ../
docker run -v /root/opentx:/opentx -e "OPENTX_VERSION_SUFFIX=2.3" -w /opentx/build-t16 opentx make firmware
```

redjacketKR, 2019-10-29:
Better yet maybe you could share the compiled firmware temporarily.

robustini, 2019-10-31:
> 
> 
> Better yet maybe you could share the compiled firmware temporarily.

There's PR by @3djc about this, we await the merge, then we can choose the BT option from Companion.

closed 2019-11-04 by 3djc

SeanTheITGuy, 2020-01-24:
> Bluetooth on T16 works really well, I use that of the X12S (the most economic, $9), but the automatically compiled firmware does not provide it, so at the moment you have to use a self compiled binaries with the BLUETOOTH option.

Would there be a way to use this functionality to piggyback a "follow me" type app's output from a phone over the transmitter's link to be used in GCS Nav mode via MSP on a quad?  Trying to find the least obtrusive way of getting this functionality, and right now it seems like building the iforce2d followme tag and controller seems like it, especially given the work he's put in to the various follow modes, like circling a target, etc.

DIY420, 2020-04-02:
Question to S.Port mode: Is is sufficient to listen to the existing S.Port stream from transmitters main board to the RF module, just inverting this stream and lead it to u360gts-RxD? Or do I have to use both data directions, u360gts-TxD for telemetry request and u360-RxD for telemetry receive (its clear that in this case I have to use the bluetooth interface usart and activate the bluetooth module in my Jumper T16, which is luckily available in opentx_T16 in the meantime).

fabi9163, 2021-11-13:
> Bluetooth on T16 works really well, I use that of the X12S (the most economic, $9), but the automatically compiled firmware does not provide it, so at the moment you have to use a self compiled binaries with the BLUETOOTH option. So since the Bluetooth works you can theoretically do everything you do on an X12S / X10, I currently only tested the TRAINER function. You can see it installed...
> 
> [image]

Any news? Which bluetooth-Module exactly did you use? I'm searching for a build in way to export telemetry from my T16 Pro v2 via bluetooth to my smartphone. solder bluetooth-module, enable bluetooth-flag in companion-app and go for it?

rotorman, 2021-11-13:
> Which bluetooth-Module exactly did you use?

This one is a solder-in module with antenna and appropriate firmware pre-loaded: https://de.aliexpress.com/item/4001192317700.html

PR #7020 (2019-11-04): Add support for BT compilation on T16
This fixes #7010