opentx/opentx #7012: OpenTx 2.3.1 Taranis X9D+ 2019 - Intermittently bad telemetry data (on both ACCESS & ACCST)

NickMilton, 2019-10-27:
**Describe the bug**
Two examples...both have reproduceable issue..

(1) Taranis X9D+ 2019 OpenTx 2.3.1 with X8R running ACCST ISRM 8ch + telemetry (has MLVSS + GPS modules)
(2) Taranis X9D+ 2019 OpenTx 2.3.1 with G-RX8 running ACCESS ISRM 8ch + telemetry (has MLVSS + Redundancy Bus)

All devices have most recent stable firmware installed. Note that Example#1 had been running reliabily for yuears on a X9D+SE 2017 radio. Example#2 is a new glider.

**To Reproduce**
Steps to reproduce the behavior:
Turn on Taranis, turn on model battery supply. Walk 5 meters away from model to avoid 'Telemetry lost' issue. Within less than 1 minute the values displayed on the Taranis telemetry screens show 'nominal values' (eg 1Amp or 12Volts) which erratically and intermittently change to enormous values  (eg 3000Amps or 23000Volts). Then all the configured alams go off (annoyingly). Obviously the redundancy bus starts sending all sorts of crazy messages like over current  - which are not true. The glider showed an altitude of over 13254160.8 meters, then moments later showed a correct height.
[Bad telemetry data logs and configs.zip](https://github.com/opentx/opentx/files/3775631/Bad.telemetry.data.logs.and.configs.zip)

**Expected behavior**
Stable and meaningful temetered data

**Screenshots**
If applicable, add screenshots to help explain your problem.

**General information**
Versions as above
In the radio configs attached example 1 is Edge 540 60CC and example 2 is Phoenix 2400

bsongis, 2019-10-28:
You will be contacted by FrSky shortly to solve this issue. If anything wrong on OpenTX side please let us know.

closed 2019-10-28 by bsongis

NickMilton, 2019-10-28:
Thanks for  your support.

To be clear I cannot say if this is an OpenTx/firmware  issue, a
configuration error or something to do with the new Taranis Plus SE 2019
hardware or other.

Can you tell me when FrSky will be on contact as I cannot fly with radio in
current state.

Appreciated.

NickMilton, 2019-10-31:
No one seems to have contacted me from FSky. I think this is an OpenTx firmware issue. Please reopen this