opentx/opentx #7014: X-Lite (Original): R9mm-ota firmware upgrade via ota fails

rickydee, 2019-10-28:
OTA firmware upgrade of r9mm-ota using x-lite (original) / R9 lite pro fails. OpenTx 2.3.1 installed on radio.

Also tested on latest stock x-lite firmware (190905) downloaded from frsky website - r9mm-ota firmware upgrade via ota appears to function correctly.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to SD-HC Card / Firmware
2. Click on R9mm-ota .frsk file
3. Select 'Flash Receiver OTA'
4. X-Lite displays 'Waiting for RX'
5. Turn on R9mm-ota receiver
6. X-Lite returns to the file menu. Binding chirp sound remains and keys do not function. Requires power cycle at this stage. 

- OpenTX V2.3.1
- Hardware TX: Frsky X-Lite Original/R9 lite pro RX: R9mm-ota

bsongis, 2019-10-28:
Would you please check this one in the nightly builds? I fixed it already

XTreM3com, 2019-11-17:
I got exactly the same Problem. Same Hardware. I tried v2.3.1, v2.3.2 and now the latest nightly (v2.3.3N45 (70a9e531)) and even there the problem still exists.
@bsongis Do you refer to https://github.com/opentx/opentx/issues/6771 with the fix you mentioned? If so, it seems to be already included in the stable v2.3.x releases (not just the nightlies). See e.g. 2.3.1 branch: https://github.com/opentx/opentx/commits/2.3.1/radio/src/gui/common/stdlcd/model_setup_pxx2.cpp

Firmware options in Companion: noheli, lua, flexr9m
Transmitter displays: crossfire, flexr9m, noheli, multimodule, lua, luac

Hope this helps... anything else I can test?

commit f630c5bc (2019-11-26): Fix #7014 (OTA RX Update)
closed 2019-11-26 by bsongis