opentx/opentx #7016: Constant low battery warning in OTX 2.3 simulator if radio set for 3S

VancouverUmbrella, 2019-10-28:
It seems the simulated radio battery voltage in Companion 2.3.1 is hardcoded to 7.3V.

At that voltage, I get a constant low battery warning in the Companion simulator because I've set my radio low voltage alarm for 11.0V (3S Lipo).

Edit: Problem did not occur in OTX 2.2.4 or earlier versions.

Thank you,
Andrew

VancouverUmbrella, 2019-10-28:
Apologies. 

This bug is a duplicate of 
https://github.com/opentx/opentx/issues/6980

And related to:
https://github.com/opentx/opentx/issues/7007

VancouverUmbrella, 2019-10-28:
I'll leave it open for now as the other bugs specifically refer to a different radio profile, X9D I think.

3djc, 2019-10-29:
Closed, replaced by new feature request : #7019

closed 2019-10-29 by 3djc