opentx/opentx #7023: Companion 2.3.1 not opening 2.0.x EEPE files correctly

mawzthefinn, 2019-10-29:
Converted a X9D+ BIN to eepe using Companion 2.0.20

[Wonder-23conversionissue.zip](https://github.com/opentx/opentx/files/3784671/Wonder-23conversionissue.zip)

Open said EEPE in Companion 2.3 results loss of mixes and other programming items
Open said EEPE in Companion 2.2 delivers correct results

The original model save BIN, the Companion 2.0 conversion to EEPE and 2.2 and 2.3 OTX's based on the EEPE are attached

robustini, 2019-10-31:
I confirm, there is this problem, I also tried with 2.2 and 2.3 and it does the same thing here with some "old" eepe.

mawzthefinn, 2022-12-14:
You need each version of companion between the radio version and 2.3.

ie if the radio is on 2.0.x, you would need Companion 2.1, companion 2.2 and Companion 2.3 all installed.

Open the eepe in companion 2.1, let it convert, save it.
Open the now converted file in companion 2.2, let it convert, save it.
now you have a file that can be safely opened in Companion 2.3 and written to the updated radio.

treadwm, 2022-12-14:
Thx for the quick answer. This workflow did occur to me. But I've been unable to find a download for the 2.2 Companion software. The [link ](https://downloads.open-tx.org/2.2/release/companion/windows/companion-windows-2.2.4.exe) is there on OpenTX but there is no file. Nor did I find it on github.

Do you know of any alternate sites for downloading it?

treadwm, 2022-12-14:
Nevermind. Companion 2.3 read the 2.1 files without issue. 
Thanks for the assist!