opentx/opentx #7027: 2.3.2N32@X9D radio reboots if Trainer Mode Master/Serial is selected

Allerhopp, 2019-10-31:
MPM FrSky RX works perfect in Jumper T16. When I tried it with X9D the radio reboots, as soon as  Trainer Mode Master/Serial is selected. No chance to select Master/Multi. No matter if internal or external RF or none of both is activated.

3djc, 2019-10-31:
Well, We will look into that, but multi trainer cannot work on X9D because sport line is shared on this radio

3djc, 2019-10-31:
You want minimum lag on your trainer, you cannot wait the XJT to poll

raphaelcoeffic, 2019-10-31:
@Allerhopp The other reason why this would not work is the internal XJT module is driving the SPORT bus in the role of a bus master. It is not foreseen to allow bus slaves to place packets on this line whenever they please.

Allerhopp, 2019-10-31:
Correct. That's why MPM would have to wait for XJT polls. 3djc has latency concerns, but it would be possible in priciple.

kilrah, 2019-10-31:
Slow. A poll only gets you 4 bytes through and you may only get a few per second.

commit 3f0ce109 (2019-11-07): Fixes #7027
closed 2019-11-07 by bsongis