opentx/opentx #7032: [Companion] Does not load configuration file with Ubuntu 18.04

OpenMik, 2019-11-01:
Companion 2.3.1 on 
Xubuntu 18.04
Mint 19

From menu setting -> import application setting   load a configuration file saved with another Pc with same version of Companion (2.3.1)
apparently everything seems to be regular but configuration remains the same
Configuration not change also after Companion restart 

attached my application settings file
[Companion 2.3.conf.zip](https://github.com/opentx/opentx/files/3798851/Companion.2.3.conf.zip)

kilrah, 2019-11-02:
What were you expe cting to see? That configuration file is bsically empty, just the window position and a radio type for the default profile.

OpenMik, 2019-11-02:
I'm sorry
I sent the wrong file
this is the right file
[Companion 2.3 settings 01-nov-19.ini.zip](https://github.com/opentx/opentx/files/3800214/Companion.2.3.settings.01-nov-19.ini.zip)

OpenMik, 2019-12-08:
Import configuration is still not working with Companion 2.3.3N62 on Ubuntu and Mint
With Windows 10 is all ok
only with a manually replacement of ~/.config/OpenTx/Companion2.3.conf is possible to import the saved configuration
what do you suggest?
Thank you

OpenMik, 2019-12-21:
Import configuration still not working on Companion 2.3.3 stable release

OpenMik, 2020-01-01:
[config.zip](https://github.com/opentx/opentx/files/4013920/config.zip)
apparently the configuration file exported (.ini) is equal to .conf file saved in /.conf/OpenTx directory
but  .ini file is not loaded by Import Application settings
if  Companion2.3.conf is renamed in Companion2.3.ini  the import Application Settings work correctly and the saved configuration is succesfully loaded

closed 2020-09-28 by stale[bot]