opentx/opentx #7034: T16 pro (internal module) D8 telemetry not function

Kelvenma, 2019-11-02:
I bind my D8RII receiver with new T16 pro at D8 mode, It bind successfully and work fine but the telemetry function not work (no RSSI, no A1, A2 can be found).

And if I use my jumper 4in1 external module bind in D8 mode, it work perfectly with all sensor works and RSSI

OpenTX version 2.3.2N26 (actually all version on 2.3 )
multiprotocol module version 1.2.1.51 and all after 
Jumper T16 pro (internal module)
Thanks!
Kelven

3djc, 2019-11-02:
It does work fine. You need to ensure you have proper telemetry from the module (your module should display it's version in the model setup internal module section

closed 2019-11-02 by 3djc

schneeg, 2019-12-17:
> It does work fine. You need to ensure you have proper telemetry from the module (your module should display it's version in the model setup internal module section

3djc, can you offer more specific information? I'm having the same issue as Kelvenma was having. D8 binding and control works fine, but no telemetry, no sensors found. What is the module version needed for 'proper telemetry'?

3djc, 2019-12-17:
First you need to ensure that your module does have telemetry return to OpenTx working : can you see the version number where youy setup the module ?

Then you don't say what you are binding to

Kelvenma, 2019-12-17:
finally I had updated the (using USB device) internal module firmware and it back to work now, just strange that it works on my external module with an older firmware (even the firmware older then the T16 pro from factory)

kilrah, 2019-12-17:
Becasue the firmware the factory flashed on their module wasn't the correct one.

schneeg, 2019-12-17:
Aah OK, that makes sense. Thank you.

@3djc: I have the same module version listed in Kelvenma's original post,
and was binding to D4R-II in D8, D8R-ii Plus in D8, and XM+ in D16. None of
them received telemetry, but my Qx7 transmitter would. I was going to try
updating the firmware on the internal module next, sounds like that's my
ticket.

Thanks for the help, guys/gals.