opentx/opentx #7035: RSSI source in Telemetry is no longer available

dposea, 2019-11-02:
I changed some option by editing RSSI, and now I get no sensors at all. RxBT and RSSI used to be always discovered. I'm not sure what to do now. Should I start a new model, restore the EEPROM?
I also no longer can select RSSI on the Inputs Source field.

Running Opentx 2.3.1 on a Taranis QX7.

dposea, 2019-11-02:
Update, I started a new model and had the same problem until I rebound the R-XSR receiver. Only default and External were available in RSSI source, but I was using the internal module. It works on default, but Internal should be a valid choice, and External really should not be unless it is enable in model setup.

closed 2020-04-30 by kilrah

al21883, 2020-09-17:
I have the same issue.
Can't select any source for RSSI on the telemetry page, except "(default)"
In my case, it's necessary to change the source, because I'm flying an old Hubsan H107D, and it only reports TRSS.
I have tried to delete the RSSI sensor that is reporting 0 but same result.

Runing 2.3.9 on a TX16S

kilrah, 2020-09-18:
This is to select the source for the built in RSSI handling, which only supports RSSI so selecting TRSS would do you no good. Just set your own alarms with special functions.