opentx/opentx #7052: [sky9x] 8th usb joystick axis is unusable

AlexDaniel, 2019-11-06:
When using transmitter as a USB joystick, the first 8 channels (CH1…CH8) of the transmitter are mapped as 8 analog joystick axis (the rest are buttons). However, both CH7 and CH8 appear as 7th channel, meaning that you cannot use all 8 channels (only 7).

```jscal -c /dev/input/js0``` reports:
```
Joystick has 7 axes and 24 buttons.
```
I think this is the problematic piece of code:
https://github.com/opentx/opentx/blob/59d3f2d16f5aae5a31bd069bec7653da3936bf87/radio/src/targets/sky9x/usb/device/hid-joystick/HIDDJoystickDriverDescriptors.c#L141-L148

Both 7th and 8th channels are `HIDGenericDesktop_SLIDER`, and should probably be different values.

Ping @AntonEvmenenko

AntonEvmenenko, 2019-11-06:
Hey!

Well, you should understand, that the above code is just a copy of similar one from joystick implementation for Taranis. Here is the original one:

https://github.com/opentx/opentx/blob/b86658acfbec753c0f48c5fb54c07174c39d99ce/radio/src/targets/common/arm/stm32/usbd_hid_joystick.c#L127-L134

Actually, now in this case we have a problem with code duplication. I plan to fix it someday (or maybe someone will do this).

About your problem: as I understand, the same problem exists for Taranis. Also, it looks like there is a PR: #6383

AlexDaniel, 2019-11-06:
Speaking of code duplication, [can we have this change too](https://github.com/opentx/opentx/pull/5825/files)? 8-bit values are kinda limiting.

AntonEvmenenko, 2019-11-06:
Exactly.

Actually, here is my plan:

As you know, there are 3 descriptors that are needed for creating a HID device:

- Device descriptor
- Configuration descriptor
- Report descriptor

Device descriptors are different for Taranis and Sky9x joystics, but configuration and report descriptors are completely the same. However, currently they are are not in sync for Taranis and Sky9x joystics. Moreover, different structures (located in different files) are responsible for these descriptors:

- For Taranis -- radio/src/targets/common/arm/stm32/usbd_hid_joystick.c
- For Sky9x -- radio/src/targets/sky9x/usb/device/hid-joystick/HIDDJoystickDriverDescriptors.c

I suggest the following: we should create file with configuration and report descriptors for USB HID joystick, and that file then should be included in code for Taranis and Sky9x joystics. As this file is not related to any architecture or pcb, in my opinion it sholud be located in "radio/src/targets/common/any/USB/HID" folder. Also, this file should contain function that forms HID report, because now there are also 2 copies of this function for Taranis and Sky9x joystics.

AlexDaniel, 2019-11-06:
FWIW, in some of the mentioned tickets there are concerns that changes to the way the joystick works can break existing setups. Just wanted to say that for 9xpro this is a non-issue, because the joystick support was merged very recently, and moreover the latest release is somewhat botched anyway because of https://github.com/opentx/opentx/issues/6914. So I think any changes to the way 9xpro joystick works can go straight in without any issues.

AlexDaniel, 2020-05-04:
Thanks bot, but it's still a bug :)

AlexDaniel, 2020-11-01:
Thanks bot, I'd still love for this to get fixed…

AlexDaniel, 2021-06-03:
But it's still an issue??

sejnub, 2021-08-28:
Hi!

I just suffered from the fact that the 8th channel is unusable. Wanted to use it on a flight simulator.

Please fix this!

AlexDaniel, 2022-04-16:
I… don't get the point of that bot 🤔 I still have that issue, even if I use workarounds