opentx/opentx #7058: X10S Express under OpenTx 2.3.2N33 doesn't handle correctly .frsk file

dev-fred, 2019-11-06:
**Describe the bug**
Under OpenTx 2.3.2N33, on X10S Express simulator or real, the SD browser doesn't identify autoupdate.frsk file as a firmware file, I have only the choice to copy/delete/rename like any file and nothing for flashing
[image]

**Expected behavior**
Under X9D+ 2019 simulator, the SD browser on my autoupdate.frsk file add the choice : Flash pwr mngt unit , exactly what I need to update the PMU of my X10S Express
[image]

**The autoupdate.frsk file**
This file is supposed to correct a PMU problem in X10S Express: external HF module is powered when USB charger is connected.
[autoupdate.zip](https://github.com/opentx/opentx/files/3816804/autoupdate.zip)

bsongis, 2019-11-07:
`HARDWARE_POWER_MANAGEMENT_UNIT` is not defined on X10 Express

bsongis, 2019-11-07:
PMU now defined on X10 Express

closed 2019-11-07 by bsongis

dev-fred, 2019-11-07:
Thanks a lot :) with this feature I should  be able to update my PMU. 
I have also activated autoupdate option and placed this file under FIRMWARE, after a reboot of the X10 I expected a flash of the PMU but nothing happened, is it also linked to this definition ? 
Just to be sure, to avoid a PMU brick, I should use this method of flash and not the browser one isn't it

3djc, 2019-11-07:
No autoupdate should be last resort, and not used unless PMU is already bricked. So from the SD manager is what you want to do

dev-fred, 2019-11-08:
OpenTx 2.3.2N36 recognizes the file autoupdate.frsk sent by Frsky but unfortunnatly this file doesn't solve my X10Express PMU problem:
The black screen Device reset ... with a progress bar  
doesn't show any progress and after a timout I got 
Erreur de mise à jour
No answer
[RTN]
This file has at least the merit of being harmless.
May be one day Frsky will produce an official patch who knows, I have sent an email and I cross my fingers.
Still Thx a lot for your fast fix

bsongis, 2019-11-08:
Did you try to enable the autoupdate on radio startup?

dev-fred, 2019-11-08:
I flashed the X10SE with the autoupdate option and I have a file autoupdate.frsk on FIRMWARE and nothing special happens during radio startup: no flash and the file is not deleted as expected. 
Have I forgotten a step?

bsongis, 2019-11-08:
OK. I am on it

dev-fred, 2019-11-08:
Don't know if it's important but under SYS/VERSION/OPTS = crossfire, multimodule, lua, luac but no autoupdate and my firmware = opentx-x10express-autoupdate-lua-fr.bin

3djc, 2019-11-08:
Target is to remove autoupdate options, it should not exist and was done to fix a one off issue at FrSky

dev-fred, 2019-11-09:
I am really grateful to you and OpenTX team, your correction **PMU flash from Horus** has made it possible to flash the PMU of my X10ES and to correct the power problem of the external HF module when you power the USB plug. The patch called autoupdate.frsk that FRSKY sent me works perfectly.
To apply it I had the autoupdate option enabled but I wonder if it is necessary because finally even after rebooting after installing the 2.3.2N37 firmware, it doesn't seem to me that this option worked, I have used the SD browser.