opentx/opentx #7059: BOOL option corrupted in Create() function of Lua widget

RC-SOAR, 2019-11-06:
I'm having a weird issue with corrupted data in a Lua widget. Minimalised code shown below. _options.UseDef_ should always be passed as 0 or 1. However, in the Create() function, it's occasionally passed as an apparently random number when a '1' is expected. The same number may appear in subsequent Create() calls. Curiously, the LS bit is correct.

I can reproduce this on two transmitters (Horus X12S and Jumper T12) and also in Companion sims for OTX 2.2 and 2.3.2 (nightly). The problem occurs after a random combination of Select/Remove/Settings operations on the widget.  Please see also attached screenshot.

```
local VERSION = "0.9"
local defaultOptions = {
	{"UseDef", BOOL, 1},
	{"BackColor", COLOR, DARKGREY},
	{"ForeColor", COLOR, WHITE}
  }

local function create(zone, options)
	print (string.format("%s, %X","-- calling create:",options.UseDef))
	return {zone=zone, options=options}
end

local function update(adpwidget, newOptions)
    adpwidget.options = newOptions
	print (string.format("%s, %X","-- calling update:",adpwidget.options.UseDef))
end

local function background(adpwidget)
end

local function refresh(adpwidget)

	local x
	local y
	local flagsColor

	lcd.drawText (40, 10, string.format ("%s%X","bool option=", adpwidget.options.UseDef))

	if adpwidget.options.UseDef==1 then
		flagsColor = 0
	else
		flagsColor = CUSTOM_COLOR
	end
end

return {
	name="allinf"..VERSION,
	options=defaultOptions,
	create=create,
	update=update,
	refresh=refresh,
	background=background
	}
```
[image]

tueddelclick, 2019-11-07:
Looks like it‘s implemented like the interpretation of bool in C where !=0 is true. 
I can confirm this problem on a Horus X10s with 2.3.1.

kilrah, 2019-11-07:
Correct.
https://github.com/opentx/opentx/issues/5264

RC-SOAR, 2019-11-07:
@kilrah Just to clarify: this is a typical sequence of the BOOL argument received in the Create() function. 

```
00000000
00000001
00000000
00000001
00000000
00000001
00000000
02404101 - first corruption
02404100 
02404101 
02404100
etc.
```

The 'bad' bits are random, in this case we had a sequence of 0240410, but it may be a different sequence on another run. The sequence is persistent between power cycles.

The LS bit is always correct - a workaround is to test using _bit32.band (option, 1)_, but that doesn't help existing scripts.

I suspect this hasn't been raised before because of its random nature, but it surely must be considered a bug.

3djc, 2019-11-08:
Clearly, we want at the very least to understand what is going on. Any guidance you can give me to try to reproduce ?

bsongis, 2019-11-08:
I take it

3djc, 2019-11-08:
Reproduced

closed 2019-11-08 by bsongis

3djc, 2019-11-08:
We have pushed a fix, HOWEVER, it will not fix existing widget instances that have possibly wrong value. Simply removing the faulty widget from screen setup and adding it again will  fix it

RC-SOAR, 2019-11-08:
Many thanks.

commit bb460089 (2019-11-08): Fixes #7059
Fixes #5264