opentx/opentx #7060: If too many sensors are available during discovery, Taranis Q X7 locks up, requires hard reboot

john-slattery, 2019-11-08:
**Describe the bug**
When setting up the sensors on my QX7, I get a notification that, "All slots full!", with the option to exit, however, despite pressing every button, the notification will not go away, which results in the radio being unusable until a power cycle is performed. 

**To Reproduce**
Steps to reproduce the behavior:
1. Have more than 40 sensors in the S.Port Chain (this is due to a bug in the EagleTree Vector's S.Port implementation, which I am contacting them about, however there is still a bug in OpenTX).  
2. Click on "Discover new sensors"
3. "All slots full!" message appears
4. Unable to dismiss message and radio must be power cycled.

**Expected behavior**
When performing this, I would expect the "All slots full!" message to go away. 

**Video demonstration of bug**
https://www.youtube.com/watch?v=5dwyO8zSVak

**General information**
 - OpenTX version 2.3.1
 - Hardware: 
   - FrSky Taranis QX7
   - EagleTree Vector (latest firmware)
   - FrSky R9M (latest FCC firmware)
   - FrSky R9 Slim+ (latest FCC firmware)

3djc, 2019-11-08:
I guess we should display the warning and stop discovery at the same time

kilrah, 2019-11-08:
As it is you should power the receiver off before dismissing the notice so it stops flooding with sensors. Auto disabling discovery is a good idea.

PR #7062 (2019-11-08): Stop sensor discovery when table is already full
This fixes #7060
closed 2019-11-08 by bsongis