opentx/opentx #7076: Taranis x9d+2019 and R9M-2019, S.Port specifications on R9M-2019 unknown

Mmarsian, 2019-11-11:
Hello All,
with X9d+ and R9m I had this solution: 
https://github.com/CrazyDude1994/android-taranis-smartport-telemetry
working, a Bluetooth module connected to smartphone telemetry app. The S.Port of the old R9M was outputting telemetry data with baudrate 57600 to the Bluetooth module. Everything worked fine.

Now I changed to X9d+SE-2019 and R9M-2019 and flashed Access (flex) firmware to the R9M-2019 and the R9MM. Control connection and Telemetry back to the Taranis are working fine, but I cannot get any useful data out of the S.Port interface on the R9M-2019. This is mainly because of the unknown specifications. Maybe the baudrate has changed or there is encryption, etc...
I would be very happy if I can get some information to bring the telemetry app back to life.

kilrah, 2019-11-12:
S-Port is not used anymore in ACCESS mode, telemetry comes back along with a bunch of other data on a 450kbaud serial link on pin 2.
https://github.com/opentx/opentx/blob/2.3/radio/src/telemetry/frsky_pxx2.cpp

closed 2019-11-12 by kilrah

skaman82, 2020-05-17:
@kilrah i have an x-lite with integrated BT Module and I am using the R9M Lite Pro with ACCESS. Having the same issue with telemetry forwarding. If I connect the crossfire module to the x-lite, I get the telemetry stream to the tracker via HM-10. With the R9M it isn't working.  Is there any workaround to get that working with R9M?

StefanKellerAC, 2020-05-29:
find my solution with an X10 and ACCESS upgrade module here:
https://www.rcgroups.com/forums/showpost.php?p=44636915&postcount=1005