opentx/opentx #7084: [Companion 2.3.2N40] - Models conversion issue from X10S Express to X12S (not respected LS/RS pots)

robustini, 2019-11-13:
As shown in the attached image the LS and RS pots during conversion from "X10S Express" to "X12S" are not respected, they switch into S3 and S4 when they should remain as they originally were.

[image]

**General information**
 - OpenTX Companion V2.3.2 N40

robustini, 2019-11-23:
@3djc can you fix it? Thanks!

3djc, 2019-11-23:
Probably not. We do not have mapping tables, that would be a dauting task to create all the best maps between all radio combinations

robustini, 2019-11-25:
Unfortunately the X12S, unlike all the other OpenTX radio compatible, has S3 and S4 pots, so I think we can fix this.
It is the only tx to have these two additional pots, so if perform a conversion to this tx it would be enough not to contemplate in any case these two pots, but only LS, RS, S1 and S2.

closed 2020-05-23 by robustini