opentx/opentx #7087: Trim for Nick-Down stucks on X9D plus 2019

sh-hedge, 2019-11-14:
My nick trim automatically goes down minimum and in debug mode I can see that it stucks.

I measured on the pins. Hardware is ok.
I have 3.2volts on the trim up pin, and only 1.6volts on the trim down pin.

Could somebody check the code that the pullups are configured correctly?

bsongis, 2019-11-14:
wow you are right

bsongis, 2019-11-15:
Would you be able to test the nightly builds? I am commiting a fix, but my radio doesn't have this issue

closed 2019-11-15 by bsongis

sh-hedge, 2019-11-15:
Yes sure, I'll test it. 
But I need some help to compile the firmware file. I do not have the software on my computer...

bsongis, 2019-11-15:
Please send me an email, and tell me which options you like in Companion, I will send a firmware so that we can have a test as quickly as possible. Thanks

commit d9333ef0 (2019-11-15): Fixes #7087