opentx/opentx #7089: Multiprotocol module fails to start ver 2.3.1

Ae5gt, 2019-11-15:
noheli option selected . Multiprotocol set to MULTI Flysky , Internal RF set to OFF, Rcvr = FS -GR3E , Default settings

Multiprotocol module fails to restart on model selection or after power cycling resulting in loss of control .

With Receiver bound and operating properly  , power on X9lite is power cycled off and then back on. Receiver flashes slowly no response. Multiprotocol option in model menu is cycle to OFF then back to MULTI  to restore control.  

Changing to a different model and then back produces the same results

Talha909, 2019-11-16:
Please update the boot loader of multi protocol

closed 2019-11-17 by kilrah

Ae5gt, 2019-11-19:
-X:~/flash-multi-0.2.2$ ./flash-multi -f
multi-stm-opentx-aetr-inv-v1.2.1.85.bin -p /dev/ttyACM0
flash-multi 0.2.2

This program is Free Software and has NO WARRANTY.
https://github.com/benlye/flash-multi/

Multi Firmware Version:    1.2.1.85 (STM32)
Expected Channel Order:    AETR
Multi Telemetry Type:      OpenTX
Invert Telemetry Enabled:  True
Flash From Radio Enabled:  True
Bootloader Enabled:        True
Serial Debug Enabled:      False

Proceed with firmware update? [Y]es or [N]o: y

Attempting USB upload using dfu-util

[1/2] Resetting to DFU mode...
./tools/64bit/maple-reset /dev/ttyACM0
maple-reset 0.1

This program is Free Sofware and has NO WARRANTY

Reset sequence sent to /dev/ttyACM0

[2/2] Writing firmware...
./tools/64bit/dfu-util -d 1eaf:0003 -a 2 -D
"multi-stm-opentx-aetr-inv-v1.2.1.85.bin" -R
dfu-util 0.7

Copyright 2005-2008 Weston Schmidt, Harald Welte and OpenMoko Inc.
Copyright 2010-2012 Tormod Volden and Stefan Schmidt
This program is Free Software and has ABSOLUTELY NO WARRANTY
Please report bugs to dfu-util@lists.gnumonks.org

Filter on vendor = 0x1eaf product = 0x0003
Opening DFU capable USB device... ID 1eaf:0003
Run-time device DFU version 0110
Found DFU: [1eaf:0003] devnum=0, cfg=1, intf=0, alt=2, name="STM32duino
bootloader v1.0  Upload to Flash 0x8002000"
Claiming USB DFU Interface...
Setting Alternate Setting #2 ...
Determining device status: state = dfuIDLE, status = 0
dfuIDLE, continuing
DFU mode device DFU version 0110
Device returned transfer size 1024
No valid DFU suffix signature
Warning: File has no DFU suffix
bytes_per_hash=1944
Copying data from PC to DFU device
Starting download: [##################################################]
finished!
state(8) = dfuMANIFEST-WAIT-RESET, status(0) = No error condition is present
Done!
Resetting USB to switch back to runtime mode

Firmware flashed successfully.

Problem still persists after reflash.  The module led shows red , uplugging
and replugging the module has no effect  led still shows red. status on
model page shows "no telemetry" , To clear the error you have to cycle the
module off and back on , status on model page then shows current version of
software .

If there is an error such as a Switch Warning , or Throttle Warning , its
almost guaranteed to cause it to lock out the module.

I am using the Vantac MPM on MULTI Flysky with Gr3e Rcvrs