opentx/opentx #7094: Para bluetooth not working on X10 Express

lthole, 2019-11-17:
When testing the Para trainer function on an X10 Express 2.3.2.N42 as master and X9D+2019 2.3.2.N43 as slave, the X10 does not discover the slave. It works the other way around. 

Steps followed:
Setting up X10 as master:
a) X9D: Create new model
b) X9D: In Hardware select Bluetooth as Trainer 
    Local Addr: 806F B096 3467
    Dist Addr: ---
    Name: xnine
c) X10: Create new model
d) X10: In Hardware select Bluetooth as Trainer 
    Local Addr: 806F B01B 6043
    Dist Addr: ---
    Name: xten
c) X9D: In Model Setup set Trainer Mode as 'Slave/Bluetooth'.
d) X10: In Model Setup set Int RF as Access, and Register and Bind an Rx4R receiver.
e) X10: In Model Setup set Trainer Mode as 'Master/Bluetooth'.
f) X10: In Model Setup scroll to 'Discover' and hit Enter. (Note it would be good to get some feedback to indicate Discover mode is active, e.g. flashing ---.)
g) Discover does not find anything.

Setting up X9D+2019 as master:
If I reverse all the above steps and swap X10 and X9D, the Discover succeeds and brings up a list of Bluetooth IDs found, including the 806F B01B 6043 from the X10. Trainer mode works as expected.

Screen grab from attempt with X10 Express as master using BlueSee on MacOS:
[image]

Screen grab from successful attempt with X9D+2019 as master using BlueSee on MacOS:
[image]

bsongis, 2019-12-03:
I just tested your setup in 2.3.3 nightlies and everything is OK.
I remember I fixed something some weeks before (X10 Discover popup), please test again with the nightlies, thanks!

closed 2019-12-03 by bsongis

lthole, 2019-12-04:
Hi Bertrand

I have updated the X10S Express and the X9D+ 2019 to nightly build 60.

Unfortunately the X10 still does not discover the X9D as slave. The other way around works.

Now here is a strange thing: after successfully testing X9D+ as master and the Express as slave, I did this:
a) With both radios on
b) Changed X9D+ from master model to slave model
c) Changed Express from slave model to master model, and went to the bottom of the Model page, and found that below the Trainer = Master/Bluetooth setting, it displayed the slave’s BT code and [Clear] next to it. This looked like the Trainer link was working, but going to the Trainer page non of the 4 stick channels were coming through. (By the way this happened to me before, but I was unsure how or when it happened, so I did not report it.)
d) I then pressed [Clear] which made [Discover] come back. Pressing [Discover] does nothing.

Regards,
Lothar

bsongis, 2019-12-04:
"Discover" should show the list of available devices.

lthole, 2019-12-04:
Yes indeed. However, with the Express as master nothing happens when I click on Discover.

lthole, 2019-12-04:
By the way, the X9D+ also sees my other BT devices, Express sees nothing.

bsongis, 2019-12-04:
I reported the issue to the person in charge of BT

makcorner, 2019-12-06:
Is there another issue opened for PARA support in X10/S Express? Will it be solved before 2.3.3 release?

3djc, 2019-12-06:
No, and this issue does not seem OpenTX related, and should be fixezd by a BT module firmware firm by FrSky, in an unknown timeframe

makcorner, 2019-12-06:
> No, and this issue does not seem OpenTX related, and should be fixed by a BT module firmware update by FrSky, in an unknown timeframe

Well, it is working well under FrOS. So seems that BT firmware is fine.

bsongis, 2019-12-06:
It works perfectly on my radio. I am afraid that the OP has an issue with his BT board (perhaps an old BT firmware)

makcorner, 2019-12-06:
I have the same issue on my X10S Express. Discovery and PARA as such works on FrOS, but not on OpenTX 2.3.2. When you hit "Discover" nothing happen, like it would not even try to scan the BT.

kilrah, 2019-12-06:
I had the same problem but it was long ago so it might not be relevant anymore, Discover would just do nothing most of the time. Sometimes the first attempt after power up would work, but wasn't nice - after pressing discover nothing visibly happened, and the popup would only come up 2-3 secs after the press. If you pressed a 2nd time in the meantime because it didn't appear to have responded then no more chance until a reboot.

makcorner, 2019-12-23:
Some update after flashing with 2.3.3:

I managed to pair Horus X10S Express as a master, with X9D+ SE 2019 using PARA trainer. As Kilrah wrote, when you hit "Discover" on X10/S/Express, it seems to be doing nothing, but after few seconds (like 6-8 sec) - and only sometimes - it shows the list of Bluetooth devices and once this happens, you can select a device and pair it. Unfortunately "sometimes" means that it works rarely, and I cannot really find any repeatable pattern. It seems that it starts to work when I was changing several times trainer mode to BT slave, BT master, jack. After one of such changes Discovery button worked, although not perfectly: with few seconds delay, without any visible reaction whatsoever (until list of devices is shown) and even than presenting only BT device addresses, not names.. Once paired, the PARA trainer is working perfectly, even after power cycle of both radios.

Should this ticket be reopened for 2.3.5? I'm happy to make some more tests, just tell me what to do.