opentx/opentx #7095: Invalid radio data after changing something.

jmceara, 2019-11-18:
**Describe the bug**
After changing something using transmitter and reboot, everything is reseted, inclusing radio data, models, calibration, etc...

**To Reproduce**
1.Create or use fresh-new downloaded SD contents using Companion
2.Move/Copy these contents to SDCard
3.Put SDCard on radio and turn it on.
4.At this time, your should do first calibration and everything should be working.
5.Reboot transmitter. Everything still working.
6.Change something in model and reboot transmitter.
7.Now, I get and "Invalid radio data" and everything is reseted to default, including calibration. Lost all models.

**Expected behavior**
Do any changes using Transmitter or Companion without any problem.

**General information**
I managed to get this error/problem 4 of 10 tests, so It's not always. I'm not sure if there's something else triggering this problem.

 - OpenTX version (e.g. 2.3.3N45)
 - Hardware Jumper T16+
 - OpenTX Companion 2.3 N45

kilrah, 2019-11-18:
Try another SD card.

3djc, 2019-11-18:
Just to confirm, is companion involved or are you doing the changes only on the radio ?

jmceara, 2019-11-18:
Is not SD Card, I already changed. Happened again now.
Radio was working 'fine'. Then, after changing between planes (in differente categories), Model name was getting white, almost could not read. Lower part of display was getting white too. Doesn't look a hardware problem, since when I change display, everything works.

Then, happened again: Model with image selected (custom picture, using the same size as original ones, 196x114 if I'm not wrong).....then lost telemetry (I have turned off receiver). At this time image of model was gone (only model image). I turned radio off and on again.....BANG! Bad radio data.

I tought that was related to companion.....but I'm not sure anymore.

jmceara, 2019-11-18:
Anyway, changed SD Cardagain. I'm using Kingston 8Gb Industrial Grade SD.

jmceara, 2019-11-19:
After switching between some pages, I managed to get 'white' on botton of screen. Note that while shutting down transmitter, thoses 'traces' around center icon get white too (I don't know how to describle that, sorry... I'm not native english speak).
I'm uploading video to youtube right now. I'll post on this comment ASAP.

3djc, 2019-11-19:
Are you using iNav lua or something similar ?

kilrah, 2019-11-19:
How did you flash firmware? Try flashing either powered off or via SD card.

jmceara, 2019-11-19:
Yes, this time I have iNAV Lua script, but was not running. But when it first happened, I had created a brand new model in Companion with brand new content (nothing else and nothing imported), then after editing in transmitter, got bad radio data after some reboots (or after changing something specific, I can't say). But that time there was no script in SD Card.

I have flashed using Companion. My transmitter had JumperTX Bootloader....now it shows HORUS Bootloader.

3djc, 2019-11-19:
The colors are probably due to iNav, thats a known issue of iNav script

Allerhopp, 2019-11-19:
Good catch btw. without iNav LUA my issue is also gone:
https://github.com/opentx/opentx/issues/6969

jmceara, 2019-11-19:
I'll try to reflash and leave without any script.

jmceara, 2019-11-19:
Ok, new problem.
I have updated to N47 (both firmware and companion) using Transmitter bootloader (not companion). Everything went fine and Radio turns on, works fine, etc,....
But now, when I plug USB cable and select "USB Storage" mode, host computer just mount one new driver (with FIRMWARE file and README) and the radio simples freeze.....nothing else works, not even power button. I need to remove battery to reset tansmitter.
When I was using N45 I could see 2 drivers in my host computer.....one with FIRMWARE and another with SD content, and transmitter still operating after remove USB cable. Now, even after unplugin USB cable, radio still freeze.

jmceara, 2019-11-19:
No success :/
All seens to be working. I have edited my model to setup timer1 and test logical switch. Then, while turning radio off, i could see 3 white dashes and no logo, black background ( totally different screen from normal shutdown). 
Note that before I try nightly build, everything as working fine with jumpertx 2.2.3, thats why I dont think its a hardware problem.

jmceara, 2019-11-19:
I may found the problem...
Look's like DragonLink is causing the problem. I have DL mounted right on the back of T16 and this is causing data corruption.......My guess is on the curring software, not on SD CArd......and when I change something in model configuration (or radio), corrupted data is saved into SD Card, causing error on thext reboot. This is only my guess......since I removed the module and tried to change model configuration and settings without any problem.
USB also started t oworking again without DL attached. I'll do more tests to make sure that this is the cause of problem.

Note that I was using exatcly this DL Module with JumperTX and no problem at all. But it's the only  thing that I can think right now for this problem.

[image]

kilrah, 2019-11-19:
Jumper are aware that their hardware is easily disturbed by high power 433MHz since they just answered another user's question by saying 433MHz systems aren't supported on the T16.

closed 2019-11-19 by kilrah

jmceara, 2019-11-20:
I didn't said before because exactly this setup was working without any issue with JumperTX. How can I imagine that one thing that was working is now causing problems? Also, OpenTX is nightly build, i mean, experimental...the chances that this is a software problem is bigger than hardware that was working, that's why I didn't mention. I'm sorry, I didn't meant to cause troubles.

Also, I know that even nightly build, looks like is not a software problem. Just to be clear.