opentx/opentx #7097: Companion 2.3.3N47 'not enough flash' error on multiple targets

lthole, 2019-11-19:
OpenTX Companion 2.3.3N47 on MacOS Mojave.
Multiple targets: 
Horus X10/X10S (crossfire, faichoice, noheli, multimodule, lua, luac)
X10 Express (crossfire, faichoice, multimode, lua, luac)
X9D+ 2019 (crossfire, faichoice, multimode, lua, luac)

When trying to download new firmware, it takes quite a while before eventually reporting an error 'Not enough flash available on this board for all the selected options'. If I repeat the download request, it immediately downloads the firmware.

3djc, 2019-11-19:
Are you trying to download firmware (ie getting it from our servers) or trying to flash ? I ask because you mention crossfire and multimode which are not choices anymore and as such do not appear.

Looking on the server, I can see opentx-x9d+2019-faichoice-lua-en, but no  x10 firmware asked today (after about 04:00 CET)

lthole, 2019-11-19:
I updated Companion, and then for each target checked for updates, downloaded the firmware and flashed the radios. The error was only experienced on X9D+ 2019 today. The X10's downloaded fine today, but also experienced these errors in the last days.

The options I listed in the issue were taken from the Version screens after successful flashing.
In Companion the only choices selected are:
Horus X10/X10S (faichoice, lua)
X10 Express (faichoice, noheli, lua)
X9D+ 2019 (faichoice, lua)
so the others are there now by default?

3djc, 2019-11-19:
For the options, yes, crossfire, multi and luac are there by default.

And sorry, I still do not understand WHERE/WHEN your error exactly is. Is that when you are trying to download the firmware (file/download/download firmware) ?

lthole, 2019-11-19:
Sorry, after the check for updates, I selected to download the firmware. At this point the progress bar pops up, and if the error is going to occur the progress bar stays at 100% for quite some time, and then the error pops up. 
Normally the progress bar goes to 100% and then resets to 0 before showing progress during the download.

3djc, 2019-11-19:
I'm again very sorry, but I have a very hard time understanding your issue

When you download a firmware that has not yet been compiled, you have the progress bars that 'cycles' while it is compiling

[image]

Then when it is done, the firmware is downloaded to the PC, and that goes with a normal bar, but takes a fraction of a second

lthole, 2019-11-19:
On MacOS, I don't get the 'cycling' progress bar during compilation, I only have a red bar at 100%. Then during download the progress bar resets to 0 and progresses normally for the download, but that can take 5-10s.

When the next update is available, I will make a video.

I also have Parallels installed, so I will use Windows Companion for the one of the updates.

3djc, 2019-11-19:
Great. If you can also record the exact time of failure, we can then try to  find traces of it in the server logs

kilrah, 2019-11-19:
You don't need to wait for an update to try again, just use the Download menu entry.

lthole, 2019-11-19:
Ok, I wasn't sure if the server does any buffering.

lthole, 2019-11-19:
I have tried on both X10 targets, and it took close to 4 minutes. I may not have waited long enough in the past, on the assumption that it would eventually compile and buffer, and download more quickly next time. Re-trying before the server was ready may have caused the issue.
However, after each error message, it definitely downloaded immediately and successfully.
I accept that I probably caused problems at your end by not just waiting long enough.

3djc, 2019-11-19:
We still need to investigate why you did not get server busiy as you should have

lthole, 2019-11-19:
Would you like me to ‘abuse’ the server by aborting and retrying too quickly?

3djc, 2019-11-19:
You can give  it a shot, I think I fixed it

lthole, 2019-11-19:
Bingo! I tried it with a standard X9D+ target, not downloaded recently. I let it run for about 60s, then cancelled, and re-tried immediately. After the error, it downloaded fine. 
Time here is 8:32pm Australian Central Daylight Time.
Couldn't attach .mov video file. Can I email it to you?

3djc, 2019-11-19:
What error did you get ? You should be getting server busy (that's what I get here on windows)

lthole, 2019-11-19:
Same as before 'Not enough flash available on this board for all the selected options'.

3djc, 2019-11-19:
OK, now we need to understand why the mac companion doesn't report the right error !

3djc, 2019-11-19:
Linux works has it should too

[image]

3djc, 2019-11-19:
We can' t reproduce, we have the same screen on mac here

lthole, 2019-11-19:
I can send you the .mov video via wetransfer if you give me an email address. Otherwise I can paste in a screenshot.

kilrah, 2019-11-19:
Upload to youtube, vimeo or equivalent and post the link.

lthole, 2019-11-20:
Here is a screenshot.

[image]

kilrah, 2019-11-20:
Yeah we know how that looks, unfortunately without context that doesn't help.
Issue is probably with one of your settings, so be sure to show companion configuration and profile settings.

lthole, 2019-11-20:
X10 Express settings below. I get the same error with Horus X10/X10S, X9D+2019, X9D+. Let me know if you want more detail.

[image]
[image]
[image]

kilrah, 2019-11-20:
Still can't reproduce or see anything that would cause this... :\

lthole, 2019-11-20:
I will ask a friend who also has a Macbook Pro (but newer) to try to reproduce the problem.

3djc, 2019-11-24:
We fixed it a couple of days ago

closed 2019-11-24 by 3djc