opentx/opentx #7110: QX7 "create a model" screen crashes.

AxleJack, 2019-11-23:
On QX7, when I create a new model, the screen goes blank and radio seems to reboot itself. The new model is created but it doesn't give the set up options, flaps channel numbers etc.

Once it gave me a syntax error number, but every other time the screen goes blank and the radio flicks off and on again.

Same firmware on the X9D+ works fine.

**General information**
 - OpenTX version 2.3.1
 - Hardware QX7

3djc, 2019-11-23:
Looks like you have an issue with your SDCard content (the new model wizard in on the sd card)

AxleJack, 2019-11-23:
Thanks 3djc. I reloaded the SD card contents, and now it works.

bsongis, 2019-11-23:
did you keep the old SD content?
the radio should not crash

closed 2019-11-23 by bsongis