opentx/opentx #7112: [Sky9x] ppm2 port not active

FXDUTRUCROSSET, 2019-11-23:
*Before submitting,  search the open issue if there is already a report of the bug*

Also often minor bugs are already fixed in nightly builds, often it is
a good idea to test (in a controlled environment like a bench setup)
with a nightly firmware to see if the bug still exists.

**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**General information**
 - OpenTX version (e.g. 2.1.23 or N123)
 - Hardware (e.g. Frsky X9D+ or 9X Pro)
 - OpenTX Companion (e.g. 2.1.23 on Ubuntu 16.04)

FXDUTRUCROSSET, 2019-11-23:
PPM2 port and TXRX2 seems not no be active on last revision 2.3 N44

blairfancy, 2020-01-26:
Hi i confirm PPM2 porrt (EXTRA-MODULE) is totaly broken. I ll try to fix that but looks complicated.

kilrah, 2020-05-01:
Duplicate 6324

closed 2020-05-01 by kilrah