opentx/opentx #7115: [OpenTX V2.3.3N54] - Jumper T16 does not go into OpenTX bootloader

robustini, 2019-11-25:
Unfortunately flashing the firmware for Jumper T16 from Companion (transmitter OFF) the tx does not access the OpenTX bootloader with the trim T1-T4.
I tried to compile the firmware locally and the same thing happens.

 - OpenTX version V2.3.3N54
 - Hardware Jumper T16
 - OpenTX Companion V2.3.3N54

3djc, 2019-11-25:
Known issue, workaround, while we fix it, turn radio on, off, then enter bootloader right away

robustini, 2019-11-25:
At the moment I stay with the HORUS bootloader, thanks anyway for the trick.

3djc, 2019-11-25:
Fixed by 3fe96bc6dd98c4da6947669bc4db1c1cf1b4aadf

closed 2019-11-25 by 3djc

robustini, 2019-12-06:
> 
> 
> Fixed by [3fe96bc](https://github.com/opentx/opentx/commit/3fe96bc6dd98c4da6947669bc4db1c1cf1b4aadf)

Not 100% fixed, on my T16 if it has been off for some time at the first power-up it does not enter in the bootloader, I have to turn it off and on again.

3djc, 2019-12-06:
Have you reflashed the bootloader or just upgraded OpenTX to RC ?

robustini, 2019-12-06:
I compiled the latest code and flashed the whole firmware from Companion.

3djc, 2019-12-06:
Radio turned off or in bootloader ?

robustini, 2019-12-06:
Turned off or the bootloader obviously would not be written.

robustini, 2019-12-06:
https://youtu.be/MXZ6Nz0WSfA

3djc, 2019-12-06:
Would you try downbloading an RC1 from us (any option), and flashing the bootloader from OTX sdcard browser with that rather than your self made version ?

robustini, 2019-12-06:
Yes, writing the bootloader this way now works.