opentx/opentx #7117: Bluetooth Trainer on T16 works as slave but not master

SilentZen, 2019-11-25:
When pairing T16 with X-Lite Pro and setting X-Lite Pro to master bluetooth trainer works but when setting T16 to master it does not.

3djc, 2019-11-25:
Please detail what doesn't work, it is a bit vague

SilentZen, 2019-11-25:
Once connected and configured as master/bt (t16) and discovered slave (xlite) I the zeros next to cal on trainer menu stay zero when moving stick on xlite.

3djc, 2019-11-25:
Not having that here at all. 

What version are you running on both ?
Do both radio say connected and display the other bt adress ?
What happens when you turn xlite off ?

kilrah, 2019-11-26:
What BT module do you have in the T16? The X-Lite Pro has a new module and if you have the older Horus one in the T16 this combo isn't compatible and only works one way.

SilentZen, 2019-11-26:
I have the X10 bluetooth module and not the X12. So this would definately be the issue.

closed 2019-11-26 by 3djc

kilrah, 2019-11-26:
To be clear what you would need is the new "para" module used in the X10 Express, not the one of the X12S.