opentx/opentx #7119: Can't extract RADIO/radio.bin

Ferfly, 2019-11-26:
When I try to open the FIRMWARE file from my radio this message appears **Can't extract RADIO/radio.bin** and I don't know what to do
 
**General information**
 - OpenTX version 2.2.4
 - FrSKY Horus X12S
 - OpenTX Companion 2.2.4

kilrah, 2019-11-26:
Becasue you don't "open" a firmware file...

closed 2019-11-26 by kilrah

Ferfly, 2019-11-26:
What? If I double click it it is opened by OpenTX...

Ferfly, 2019-11-26:
> Becasue you don't "open" a firmware file...

What? If I double click it it is opened by OpenTX...

kilrah, 2019-11-26:
Opentx registers itself for .bin files, because the eeprom files for some OpenTX radios are saved as .bin files. 

Those can be opened and edited, but for firmware it's irrelevant.