opentx/opentx #7125: backup radio /write backup to radio not implemented?

Tekniqueman, 2019-11-28:
backup radio not yet implemented and write backup to radio not yet implemented is the errors you get when trying to use the menus "Write backup to radio" and "Backup radio to file"

**To Reproduce**
Steps to reproduce the behavior:
1. Go to 'Read/Write'
2. Click on 'Write backup to radio' or 'Backup radio to file'
3. and the error 'not yet implemented' shows up

**Expected behavior**
being able to save radio.

**General information**
 - OpenTX version  2.3.3N55
 - Hardware Jumper T16
 - OpenTX Companion 2.3.3N55
[image]

3djc, 2019-11-28:
You can save your radio as an otx file, that’s how it should be done now

closed 2019-11-28 by 3djc

Tekniqueman, 2019-11-28:
Okay and that is not only for the Jumper T16? So both radio and models is saved in one file now?

Venlig hilsen
Villy Feltmann

[cid:93C05FAB-C27F-4C0F-B204-FE72EA6DA2F9]

Tlf. +45 26 13 72 52
Gentoftegade 54, 2.th.
2820 Gentofte

Tænk på miljøet, før du printer denne e-mail.
Think about the environment before printing this email.
I forbindelse med persondataloven af 25/5 2018 gør jeg opmærksom på, at Din Digitale Vicevært opbevarer følgende oplysninger om vores kunder/leverandører:

  *   Navn
  *   Adresse
  *   E-mail
  *   Telefonnummer
  *   CVR-nummer
  *   Relevante betalingsoplysninger
Hvis du som kunde fortsat ønsker, at modtage dine faktura via mail, skal du ikke foretage dig noget, og du giver os herved samtykke til, at opbevare ovenstående oplysninger
Ønsker du, at blive slettet fra vores kundekartotek, skal du venligst gøre opmærksom på dette via mail. Denne rettighed kan du til enhver tid benytte dig af.

kilrah, 2019-11-28:
Backup Radio To file and the reciprocal are only used for the radios that use an EEPROM, not those that use the SD card.

Tekniqueman, 2019-11-28:
Ah, that make sense. Thanks for the explanation.
Venlig hilsen
Villy Feltmann

Tlf. +45 26 13 72 52
Gentoftegade 54, 2.th.
2820 Gentofte

Tænk på miljøet, før du printer denne e-mail.
Think about the environment before printing this email.