opentx/opentx #7127: lcd.drawFilledRectangle doesn't support a negative width/height

LapinFou, 2019-11-28:
Hi folks,

I don't know if this is attended or not, however I got an issue when drawing a white rectangle.
Test have been done on the simulator 2.2.4 on Win 10 Pro

This does work:
`lcd.drawFilledRectangle(100, 100, 40, 40, SOLID + WHITE)`

This doesn't work (negative height):
`lcd.drawFilledRectangle(100, 100, 40, -40, SOLID + WHITE)`

Companion does crash (negative weight):
`lcd.drawFilledRectangle(100, 100, -40, 40, SOLID + WHITE)`

3djc, 2019-11-28:
Clearly not intended to support negative values. Probably can improve the doc to mention it

LapinFou, 2019-11-28:
I need to cross-check, however I pretty sure that negative height does work on the 212x64/128x64
See my script here (Line:192, the variable **hauteurAlt** is a positive value):
https://github.com/LapinFou/AltGra/blob/master/X7-X9/AltGra.lua

This why it takes me sometimes to understand what was wrong in the Horus Widget.

closed 2020-08-24 by stale[bot]