opentx/opentx #7129: X9E V2.2.4 Wrong Alert -Modell still powered

Pix4hawk, 2019-11-28:
After Update to 2.2.4 and SD-Card flashing to 2.2V0021, I get the wrong Alert when I shut down the X9E.
Regards
Mike

kilrah, 2019-11-29:
Which wrong alert? Can you clarify?

3djc, 2019-11-29:
Well,  model still powered alert means you are still receiving telemetry from your model when you try to turn off the radio.

Pix4hawk, 2019-11-29:
@3djc I know the message and it is helpful if you forget to switch off the model first. 
Since the update get this message even if the model was first turned off

3djc, 2019-11-29:
Could you detail your module and settings then ? I'm flying with X9E too and I do not have that issue

Pix4hawk, 2019-11-29:
Model (TinyHawk-S) D8 CH1-8 is selected. I'll try it later with another Model on my X9E.

Pix4hawk, 2019-11-29:
same issue with other Models

lepicfoto, 2019-12-05:
He he I also fly x9E and it does it all the time. Even when I got the receiver in hand and the function is off the menu, it's actually pretty annoying and it gives some giggle in the club "his gear is FUBAR" is a recurring thing right now, really funny not :-(

3djc, 2019-12-05:
Have you tried 2.3.x ?

kilrah, 2019-12-06:
You probably have selected the "shutdownconfirm" option.

Pix4hawk, 2019-12-06:
The shutdownconfirm is not the same Alert !
This message comes when the power button (no switch) is pressed longer to prevent accidental shutdown.
Both alarms have worked before V2.2.4

3djc, 2019-12-06:
Yes, there is an inconsistency. I guess we will make it so that shutdown confirm option disables completely RSSI check (including the setting), but that won't happen in 2.2 branch, maybe not even in time for 2.3.3

3djc, 2019-12-06:
Scrap that, I found a nice way to integrate both, preparing a PR for 2.3 branch

kilrah, 2019-12-06:
I would personally remove the shutdownconfirm option entirely.

RSSI check provides what it was initially meant for, and it makes no sense to have it only for x9e when there are now a dozen radios with push power button...

closed 2019-12-06 by bsongis

3djc, 2019-12-06:
A fix has been merged into 2.3.3 for next release

PR #7157 (2019-12-06): Better implementation of shutdown confirmation
This fixes #7129
commit 7a8e0dcc (2019-12-06): Better implementation of shutdown confirmation (#7157)

This fixes #7129