opentx/opentx #7130: idea or doability for a memory of frequency fine tuning once for all for each brand of receiver

FXDUTRUCROSSET, 2019-11-28:
would it be possible when using multiprotocol with opentx t create a sub menu or meory to store once for all the receivers of a brand 
or once the first model of a protocol, creating a memory or the fine frequency setting so next receivers bound or created in the next future should already be affected with this setting and would be bound correctly?
thx

kilrah, 2019-11-29:
Once you have found them you should hardcde them in the multi firmware, there's a facility for that.

closed 2019-11-29 by kilrah

FXDUTRUCROSSET, 2019-12-01:
André, i know you're right but it's far less user-friendly than an option for each user in companion, that would remain once set up regardless updates of new versions in preferences menu for instance