opentx/opentx #7131: t16 internal more external multiprotocol module trainer multi and other brand frsky channels offsets

FXDUTRUCROSSET, 2019-11-28:
i tried last week multi trainer with various couples of radios
if T16 master and diy2560 or sky9x as slave it seems the channels between master and slave are offset and the range of output is reduced drastically

raphaelcoeffic, 2019-11-30:
Strange, some things have already been fixed in this direction. Could you please tell us more on the setup used? Which special function used, protocols, etc?

FXDUTRUCROSSET, 2019-12-01:
My setup was t16 master/multi  with internal module bound to d8 receiver, external module jp4in 1 bound in d8 as a receiver and  sky9x v3 with xrange multi module as slave.

3djc, 2019-12-01:
What is you special function on the T16 to activate trainer ?

FXDUTRUCROSSET, 2019-12-01:
I think it was trainer channs since I didn't see any other Evocating option. I made update so I will remake full test with other transmitters there was no issue with frsky xlite. And drift with sky9x and mega 2560 based ones. I will make also a try with er9x multi avr to be complete

3djc, 2019-12-01:
trainer chans woud work as you describe it, so please try trainer -- or trainer sticks depending on your version

FXDUTRUCROSSET, 2019-12-01:
After a complete check with xlite( internal module whatever mode used or external multi module in frsky every mode( not in lbt) sky9x and even er9x with multi

FXDUTRUCROSSET, 2019-12-01:
I tried also many custom diy mega 2560 based transmitters with dht frsky transceivers and cc2500.  With nightlies n56 of opentx and revision 47 of multi module all is working perfectly without any drift or range issue both in stick and Chans modes of trainer mode. Just transmission lost on the benches, too close maybe

closed 2019-12-01 by kilrah