opentx/opentx #7132: [Horus Widget] How to init a default sensor?

LapinFou, 2019-11-29:
Hi folks,

I'm working on my first Horus Widget.
I would like to auto-select the sensor "Alt". If it doesn't exist, I want to setup the default sensor to 249.
I tried this code:
```
local defaultOptions = {
  { "Capteur", SOURCE, (getFieldInfo("Alt") or {id=249}).id },
  { "Auto",    BOOL,   1 },
  { "Inter",   SOURCE, 110 },
  { "Plein",   BOOL,   1 }
}
```
Unfortunately, it doesn't work. 249 is always selected despite the fact that a sensor "Alt" does exist.
It looks like the **getFieldInfo** doesn't return any valid data when the Widget is initialized.
Here is my question: is it possible to achieve what I'm looking for?
If yes, what will be the correct way?

Thanks.
Seb

lshems, 2020-05-27:
Relevant

offer-shmuely, 2020-07-31:
There is a workaround, you can look at the BattCheck widget

on the option, use '0', it will show the '---'
```
local _options = {
  { "Sensor", SOURCE, 0 }, -- default to 'Cels'
   .
   .
  }
```

but later in the create() func, replace it with your default:
```
local function create(zone, options)
.
.
.

  -- use default if user did not set, So widget is operational on "select widget"
  if wgt.options.Sensor == 0 then
    wgt.options.Sensor = "Cels"
  end
```

closed 2021-06-04 by stale[bot]