opentx/opentx #7134: Add GPS to jumper T16

Stinky80, 2019-11-29:
Add hardware support for the GPS module for Jumper T16 on debug port serial for use on horus x12s
on the board of the T16 humper there is an excellent way to connect the module (u-block)

Guillesan, 2019-12-03:
I am interested in this gps connection. Could you tell me where you would connect? . I guess in a uart of the plate, could this uart also be used to remove telemetry, as does the plate of the Taranis 9XE or others?

Stinky80, 2019-12-03:
there is a debug port on the board - usart3. I think to reassign it a little problem. there is still free i2c port for using 6dof axis while I have a bad code .. 

sorry for my English... google translate..

Stinky80, 2019-12-04:
Where else should I edit the text for GPS support?

Hal.h
// GPS
#if defined(PCBX12S)
  #define GPS_RCC_AHB1Periph            RCC_AHB1Periph_GPIOA
  #define GPS_RCC_APB1Periph            RCC_APB1Periph_UART4
  #define GPS_USART                     UART4
  #define GPS_GPIO_AF                   GPIO_AF_UART4
  #define GPS_USART_IRQn                UART4_IRQn
  #define GPS_USART_IRQHandler          UART4_IRQHandler
  #define GPS_UART_GPIO                 GPIOA
  #define GPS_TX_GPIO_PIN               GPIO_Pin_0 // PA.00
  #define GPS_RX_GPIO_PIN               GPIO_Pin_1 // PA.01
  #define GPS_TX_GPIO_PinSource         GPIO_PinSource0
  #define GPS_RX_GPIO_PinSource         GPIO_PinSource1
#elif defined (RADIO_T16)
  #define GPS_RCC_AHB1Periph            RCC_AHB1Periph_GPIOB
  #define GPS_RCC_APB1Periph            RCC_APB1Periph_USART3
  #define GPS_USART                     USART3
  #define GPS_GPIO_AF                   GPIO_AF_USART3
  #define GPS_USART_IRQn                UART3_IRQn
  #define GPS_USART_IRQHandler          UART3_IRQHandler
  #define GPS_UART_GPIO                 GPIOB
  #define GPS_TX_GPIO_PIN               GPIO_Pin_10 // PB.10
  #define GPS_RX_GPIO_PIN               GPIO_Pin_11 // PB.11
  #define GPS_TX_GPIO_PinSource         GPIO_PinSource10
  #define GPS_RX_GPIO_PinSource         GPIO_PinSource11
#else
  #define GPS_RCC_AHB1Periph            0
  #define GPS_RCC_APB1Periph            0
#endif

#endif // _HAL_H_

Guillesan, 2019-12-04:
My intention is not to mount GPS on the Jumper T16, but if I am interested in taking direct telemetry, knowing that there is a uart4, my question is whether it could be used for that purpose. I don't know where that uart is, would you be so kind as to take a picture of where you are on the board? Thank you

Stinky80, 2019-12-04:
[image]
Here is the port. I checked the data table gpioB pb.10, pb11

Guillesan, 2019-12-04:
Correct and thank you very much. I had the presumption that these connections were. Now you confirm it to me, I will see if any "guru" can tell me if that uart3 can be used to extract telemetry. Thank you very much.

Guillesan, 2019-12-05:
I understand that the uart3 that you intend to use for the GPS is originally for "debug", would it be possible to use it for telemetry output? As the 3djc partner explains to me here https://github.com/opentx/opentx/issues/7150 there is no free uart on the board. I am in doubt, could you clarify me.

OpenUAS, 2020-01-03:
If there is an I2C port on theT16 (There should be since the STM32 has it), an e.g. uBlox M8N module has I2C output one could use that. Same goes for a Wifi module eg. a bitbanged I2C on a ESP8266 via eg. Arduno ESP8266 transparent serial stream solution.

Stinky80, 2020-01-03:
Думаю к i2c можно включить акселерометр , гироскоп

carmanuelz, 2020-08-03:
> I understand that the uart3 that you intend to use for the GPS is originally for "debug", would it be possible to use it for telemetry output? As the 3djc partner explains to me here #7150 there is no free uart on the board. I am in doubt, could you clarify me.

I would like to know how I managed to use the uart in question to extract telemetry, I intend to use an esp32 to send telemetry to an http API and simultaneously via bluetooth to an app

Stinky80, 2020-08-06:
почему Мы считаем что debug (usart3) занят на этой плате? больше на нем ничего нет кроме debug может сделать его для gps port?  или aux
и почему нет возможности например выбрать bluetooth/gps  как на tx16s? (usart6)

Stinky80, 2020-08-06:
изучил  принципиальную схему jumper t16 (удалось найти) usart 3 свободен.  могу поделиться схемой кому надо.

Guillesan, 2020-08-06:
I would like to have that scheme of the t16.  thank you

Stinky80, 2020-08-06:
[T16 Schematic.pdf](https://github.com/opentx/opentx/files/5035629/T16.Schematic.pdf)

Guillesan, 2020-08-06:
many thanks.

rotorman, 2021-02-20:
@Stinky80 would you have a PDF of the 6-pos to analog schematic as well?

closed 2022-04-17 by stale[bot]

rotorman, 2022-04-28:
For all Jumper T16 users longing for radio GPS support, see: https://github.com/EdgeTX/edgetx/pull/1915