opentx/opentx #7137: Derived model

padcom, 2019-11-30:
*Before suggesting a new feature here, please search the open issue if there is already a request for the feature*

**Is your feature request related to a problem? Please describe.**
When running a fleet of models that are similar (for example a series of quadcopters) most of them have a similar setup. For example the stick controls are the same, the voice commands are the same, switches are usually exactly the same. Making adjustments to one configuration of one model is easy enough but changing it for the entire fleet is not only very cumbersome but also extremely error prone.

**Describe the solution you'd like**
It'd be absolutely amazing if I could setup a base configuration with all the inputs, channels, special functions and what not and to be able to override per-model things like global settings, global variables and telemetry.

That way I could have one base configuration for quadcopters and one for planes and I would be able to derive all my models from those two base configs allowing the logs to have meaningful names and being able to induce changes to all my planes and quads.

I believe one level of linking between models would be totally fine and enough. Going with a parent/parent/parent style of configuration could be very confusing.

There need to be 3 things available from the user interface:
- creating a derived model
- resetting a particular setting to parent model value (essentially clearing the override)
- display of some sort of indicator that the visible value comes from the parent configuration (a dot or star somewhere in the first line of the display would be good enough - it doesn't have to be per setting cluttering the screen for no apparent reason).

**Describe alternatives you've considered**
The alternative is what we have at the moment which is to have a set of independent configurations and to introduce changes to each and every one of them which is a pain in the neck for more than a couple of models.

lshems, 2019-12-09:
Better boost the lua API to get a more complete access to settings. This would allow for easy changes as well, for other usecases

closed 2020-09-04 by stale[bot]

padcom, 2020-09-16:
It's a pity it has not been considered :(