opentx/opentx #7138: Simulator - cant open Mix Lua script

darvin, 2019-11-30:
Open firmware simulator

SD Card path set correctly

Choosing data source - SD Card

in SD Card path /SCRIPTS/MIXES/throttle.lua

in simulator - open Model settings/custom scripts

trying to create new - throttle.lua is not there, nothing is there.

Horus 12s simulator

kilrah, 2019-11-30:
Wildcard issue it seems

```
convertToSimuPath(): /SCRIPTS/MIXES/.lua -> C:/Users/me/Documents/OpenTX/sdcard-480x272-2.3V0022/SCRIPTS/MIXES/.lua
findTrueFileName(C:/Users/me/Documents/OpenTX/sdcard-480x272-2.3V0022/SCRIPTS/MIXES/.lua)
	not found```

darvin, 2019-11-30:
@kilrah whats workaround?

kilrah, 2019-12-01:
Do it on the radio?

3djc, 2019-12-02:
throttle is 8 chars, when only 6 are allowed, reduce name lenght to 6 and it works

closed 2019-12-02 by 3djc