opentx/opentx #7139: Control failure in range test mode

mpjf01, 2019-12-01:
Since the upgrade to OpenTx 2.3 I have experienced a number of control failures when range testing. RSSI readings are OK, no warnings, but there is no movement of aircraft control surfaces. This has happened to me with a X9D transmitter and a G-RX8/XM+ receiver combination. After several switches in an out of RT mode I eventually got control surface movement.

On RC Groups two other reports are documented, with X9D+ and X8R and X9D+ and X10 with several receivers. (See OpenTx discussion thread https://www.rcgroups.com/forums/showpost.php?p=43266039&postcount=1352 and following posts).

KevinFairgrieve, 2019-12-02:
I have also encountered this issue.

Horus X12S OpenTX 2.3.2

Range check mode, walk away from model, no control. Walk back to model (still in range check mode"  and control is regained, walk away from model and normal range check works OK. Will try to capture this on video tomorrow.

3djc, 2019-12-02:
What @KevinFairgrieve is saying a rf issue. Not an OpenTX one, if you confirm there no interaction on transmitter, just by getting closer to the model.

mpjf01, 2019-12-03:
If that's was the case what is the reason that it has just started happening since the OTX upgrade? Why is there no control when the RSSI reading is way above alarm level?

3djc, 2019-12-03:
There could be different issue at work, but if control is regained without touching the radio, just by walking closer, that strongly suggest RF.

Your seems different, we need a way to replicate somehow. Did over 50+ range test with x7 yesterday  (the closesst I have from x9d), did not fail once. But you also suggest two receiver in redundancy, thats another layer of complexity

KevinFairgrieve, 2019-12-03:
I would normally agree that this is a RF issue. However it started the day after I updated to 2.3.2

I also get good rssi numbers and no control. This is happening on 2  Horus X12S TX. One I could understand, but two! Coincidence?

3djc, 2019-12-03:
We are investigating it, I was able to reproduce

3djc, 2019-12-03:
We found the source, should be fixed in next nightly

PR #7143 (2019-12-03): Range test fix (it sent only the upper or lower channels, but not both)
This fixes #7139
closed 2019-12-03 by bsongis