opentx/opentx #7141: Telemetry simulator with Companion

aviat40, 2019-12-01:
Hello,

I used to check my Lua widget with the telemetry simulator from companion but, 
from the lastest 2.3.x  version, with the same use of the software from 2.2.4, it is impossible after setting a value in the telemetry screen simulator, to check the value in the Lua widget.

Any idea, I have intalled the nightly version also to check and it is the same. The change seems to be from 2.2.4 to 2.3.x

Thank you if somebody else can try

3djc, 2019-12-01:
Have you tried rediscovering your sensors, and reaffecting thoise to the widgets ?

aviat40, 2019-12-01:
Yes of course. But, I just checked on the radio for example, the RSSI and BtRx are as ID 248 ( I asked
you previously how this is coded ) and after backup the radio with companion, on the telemetry screen with companion, the ID is 25 on the model telemtry screen, and in the telemetry simulator within companion, it is 248 ...
So i think there is a bug somewhere with the new ID between COmpanion and Opent TX

kilrah, 2019-12-01:
Duplicate of #6862

closed 2019-12-01 by kilrah

aviat40, 2019-12-01:
Thanks, Kilrah, I tried to check before in the list but !!
So the bug is known